pklaus/brother_ql #2: [Brother QL-720NW] Red flashing while printing

medouederni, 2016-02-13:
I think I didn't get the right resolution/size for the png file to convert.
Could you please provide a sample png file for a demo.

pklaus, 2016-02-13:
What label type and size do you use?

Here is my example file: 720x151_monochrome.png
[image]

medouederni, 2016-02-13:
I used 62mm label.

pklaus, 2016-02-13:
The manual states this about a flashing red LED:

> Indicates one of the following errors.
> Red lamp flashes in 1.0 second intervals:
> • End of DK Roll
> • Incorrect DK Roll installed
> • Feed error
> • No DK Roll
> • Transmission error
> • The roll cover is opened (While running)
> • Data image deployment error
> • Access point search/connection error
> Red lamp flashes 0.9 seconds
>  OFF, 0.3 seconds ON, 0.3 
> seconds OFF, and then 0.3 seconds ON:
> • Cutter error
> Red lamp flashes in 0.3 second intervals:
> •EEPROM error

Usually it's a problem with the commands in the protocol it such an error occurs.

I do not own the QL-720NW myself so I cannot debug it myself. You could try the following though:

* Run a modified version of brother_ql_create: Save the raw version of [create.py](https://github.com/pklaus/brother_ql/blob/master/brother_ql/create.py) into a folder on the Raspberry Pi 2, change it, and run this instead of brother_ql_create. (They are the same.)
  What you would want to modify are the [setup commands](https://github.com/pklaus/brother_ql/blob/master/brother_ql/create.py#L56-L67). You could comment out those function calls for example, create the .bin file again and send it to the printer to see if that helps.
* Run the vendor software to print a label and capture the network packets with `tcpdump` or Wireshark. Send the .pcap to me to let me inspect it.

I'm sure we will sort it out.

FYI: Starting tomorrow I'll attend a conference (Sunday to Sunday), so my answers might be slightly delayed.

medouederni, 2016-02-15:
I commented the line 53: qlr.set_mode(), and transferred the bin file to the printer via network.
I got a green light flashing, that means data transfer but without printing result.
Btw I got the same old behavior with the usb transfer mode.

medouederni, 2016-02-15:
Using this project: https://github.com/nametacker/brother-label
I got an infinite printer rolling without any printing:

`root@raspberrypi:/home/pi/brother-label/brother-label# ./printlabel.py
DEBUG:brother.printer:Connected to 192.168.1.65:9100
DEBUG:PIL.PngImagePlugin:STREAM IHDR 16 13
DEBUG:PIL.PngImagePlugin:STREAM PLTE 41 6
DEBUG:PIL.PngImagePlugin:STREAM bKGD 59 1
DEBUG:PIL.PngImagePlugin:bKGD 59 1 (unknown)
DEBUG:PIL.PngImagePlugin:STREAM pHYs 72 9
DEBUG:PIL.PngImagePlugin:STREAM tIME 93 7
DEBUG:PIL.PngImagePlugin:tIME 93 7 (unknown)
DEBUG:PIL.PngImagePlugin:STREAM iTXt 112 29
DEBUG:PIL.PngImagePlugin:STREAM IDAT 153 265
DEBUG:brother.task.label:40 lines of data
DEBUG:brother.printer:Sent: 0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
DEBUG:brother.printer:Sent: 1b40
DEBUG:brother.printer:Sent: 1b696101
DEBUG:brother.printer:Sent: 1b697a860a3e1d000100000000
DEBUG:brother.printer:Sent: 1b694d40
DEBUG:brother.printer:Sent: 1b694101
DEBUG:brother.printer:Sent: 1b6908
DEBUG:brother.printer:Sent: 1b696423
DEBUG:brother.printer:Sent: 1b694d00
DEBUG:brother.printer:Sent: ffffffffff8000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000ffffffffff
DEBUG:brother.printer:Sent: 80200802008000000000000000000000000000000000000000000000000000000000000000000000000000000ff80000000000000000000000000000000000000000000000000000000000000000000000000000008020080200
DEBUG:brother.printer:Sent: 80210bc238800000000000000000000000000000000000000000000000000000000000000000000000000000788f0000000000000000000000000000000000000000000000000000000000000000000000000000008020080200
DEBUG:brother.printer:Sent: 80238a2244800000000000000000000000000000000000000000000000000000000000000000000000000001c081c000000000000000000000000000000000000000000000000000000000000000000000000000008020080200
DEBUG:brother.printer:Sent: 80228a228080000000000000000000000000000000000000000000000000000000000000000000000000000700807000000000000000000000000000000000000000000000000000000000000000000000000000008020080200
DEBUG:brother.printer:Sent: 8026cbc28080000000000000000000000000000000000000000000000000000000000000000000000000000c00801800000000000000000000000000000000000000000000000000000000000000000000000000008020080200
DEBUG:brother.printer:Sent: 8027ca228080000000000000000000000000000000000000000000000000000000000000000000000000001800800c00000000000000000000000000000000000000000000000000000000000000000000000000008020080200
DEBUG:brother.printer:Sent: 802c6a224480000000000000000000000000000000000000000000000000000000000000000000000000003000800600000000000000000000000000000000000000000000000000000000000000000000000000008020080200
DEBUG:brother.printer:Sent: 80282bc23880000000000000000000000000000000000000000000000000000000000000000000000000006000800300000000000000000000000000000000000000000000000000000000000000000000000000008020080200
DEBUG:brother.printer:Sent: 80200802008000000000000000000000000000000000000000000000000000000000000000000000000000c000800180000000000000000000000000000000000000000000000000000000000000000000000000008020080200
DEBUG:brother.printer:Sent: ffffffffff8000000000000000000000000000000000000000000000000000000000000000000000000000800080008000000000000000000000000000000000000000000000000000000000000000000000000000ffffffffff
DEBUG:brother.printer:Sent: 8020080200800000000000000000000000000000000000000000000000000000000000000000000000000180008000c0000000000000000000000000000000000000000000000000000000000000000000000000008020080200
DEBUG:brother.printer:Sent: 9c2008020080000000000000000000000000000000000000000000000000000000000000000000000000010000800040000000000000000000000000000000000000000000000000000000000000000000000000008020080200
DEBUG:brother.printer:Sent: 842008020080000000000000000000000000000000000000000000000000000000000000000000000000030000800060000000000000000000000000000000000000000000000000000000000000000000000000008020080200
DEBUG:brother.printer:Sent: 842008020080000000000000000000000000000000000000000000000000000000000000000000000000020000800020000000000000000000000000000000000000000000000000000000000000000000000000008020080200
DEBUG:brother.printer:Sent: 842008020080000000000000000000000000000000000000000000000000000000000000000000000000020000800020000000000000000000000000000000000000000000000000000000000000000000000000008020080200
DEBUG:brother.printer:Sent: 842008020080000000000000000000000000000000000000000000000000000000000000000000000000060000800030000000000000000000000000000000000000000000000000000000000000000000000000008020080200
DEBUG:brother.printer:Sent: 842008020080000000000000000000000000000000000000000000000000000000000000000000000000040000800010000000000000000000000000000000000000000000000000000000000000000000000000008020080200
DEBUG:brother.printer:Sent: 9f2008020080000000000000000000000000000000000000000000000000000000000000000000000000040000800010000000000000000000000000000000000000000000000000000000000000000000000000008020080200
[... cut ...]
DEBUG:brother.printer:Sent: 9120080200800000000000000000000000000000000000000000000000000000000000000000000000000001c081c000000000000000000000000000000000000000000000000000000000000000000000000000008020080200
DEBUG:brother.printer:Sent: 8e20080200800000000000000000000000000000000000000000000000000000000000000000000000000000788f0000000000000000000000000000000000000000000000000000000000000000000000000000008020080200
DEBUG:brother.printer:Sent: 80200802008000000000000000000000000000000000000000000000000000000000000000000000000000000ff80000000000000000000000000000000000000000000000000000000000000000000000000000008020080200
DEBUG:brother.printer:Sent: 1a`

medouederni, 2016-02-16:
Finally I got the expected result with this project: https://github.com/chaosdorf/labello
But it is text only.

pklaus, 2016-02-29:
Hi medouederni! I'm back from my business trips. The "labello" project cannot work for graphics output since the library they are using ([python-brotherprint](https://github.com/fozzle/python-brotherprint)) states that they only support the text-only ESC/P and template mode.
I would like to find the problem in my **brother_ql** code and fix this with and for you. I propose two ways of proceeding: Either you send me the QL-720NW to Germany, I debug and fix the software, and then I send it back to you (~ a week later); or you try to print again via USB, reading back the error message - this isn't implemented so far but can be done (only supported via USB).

gregsadetsky, 2016-07-12:
Hi,

I'm having the same problem (following the example in the main project README) with a QL-570 -- the red light flashes when I `cat` my binary file to the USB interface.

Also, brother_ql_create complains twice that:

`WARNING:brother_ql.raster:Trying to switch the operating mode on a printer that doesn't support the command.`

however, on this [spreadsheet from Brother](http://download.brother.com/pub/com/dev/pdf/label_printer_dev_map.pdf), it seems like the QL-570 should support the 'raster' mode.

Any help would be appreciated! :-)

Thanks

---

@pklaus you mention that it's possible read back the error message via USB. Would you mind detailing how this can be done? Thanks (I'm on Debian if that helps)

pklaus, 2016-07-21:
Hi @gregsadetsky, The QL-570 definitly supports the raster mode. But not every printer understands the opcode to switch into a different mode (including the raster mode). I was working on the package a lot in the last days. Please update to the latest version and try again. If you're still having trouble, which I expect, you can try the new `brother_ql_debug` tool as explained in the README now. It reads the status information back from the device and you can see after which opcode/instruction you got the error. Please post the output here. Thanks

pklaus, 2016-11-10:
@gregsadetsky It seems there was a problem concerning the QL-570 in the code. See issue #7. Please try again and report any further problems there.

pklaus, 2016-11-10:
@medouederni Is there still an interest from your side to make the code work with the QL-720NW?

gregsadetsky, 2016-11-10:
@pklaus apologies Philipp, I won't have access to the printer for some time. Best of luck!

pklaus, 2016-11-20:
So, we now have a QL-720NW at the university and I can say that it works perfectly with brother_ql. Just be sure to call `brother_ql_create` with the `--model QL-720NW` so it can adapt to the particularities of that model. I'm closing this issue now.
Anyone else seeing a problem with the QL-720NW: please open a new issue with the image file you want to print.

closed 2016-11-20 by pklaus