pklaus/brother_ql #3: No backend avaiable using pyUsb on Macos (10.11 & 10.12beta)

enagra, 2016-09-03:
System Mac OS X (10.11)
Printer QL-570 

I followed the final instructions as follows,

`pip install pyusb`

And run: 
```
brother_ql_print 720x151_monochrome.bin usb://0x04f9:0x2028

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.5/bin/brother_ql_print", line 9, in <module>
    load_entry_point('brother-ql==0.6.dev0', 'console_scripts', 'brother_ql_print')()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/brother_ql/brother_ql_print.py", line 67, in main
    printer = BrotherQLBackend(string_descr)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/brother_ql/backends/pyusb.py", line 75, in __init__
    for result in list_available_devices():
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/brother_ql/backends/pyusb.py", line 43, in list_available_devices
    printers = usb.core.find(find_all=1, custom_match=find_class(7), idVendor=0x04f9)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/usb/core.py", line 1263, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
```

Is it possible to find an alternative not using pyusb in macos with this printer model?
[Bug on pyUsb](https://github.com/walac/pyusb/issues/120)

pklaus, 2016-09-03:
Thanks for the report. You are just missing a PyUSB backend such as *libusb*. My recommended way is to install it is via Homebrew: `brew imstall libusb`. Let me know if you need more help. I will adjust the instructions accordingly.

enagra, 2016-10-08:
Great! I am so sorry for the long delay for the response. I'll close this issue but I'll open a new because I have the problem caused by red light when printing.
Thank you for your work

closed 2016-10-08 by enagra

kevinsimper, 2018-06-19:
Non misspelled command 😅
```
brew install libusb
```