pklaus/brother_ql #6: Error brother_ql_analyse

PanderMusubi, 2016-09-20:
Please fix the following error for  brother_ql_analyse

$ brother_ql_analyse test.bin --loglevel DEBUG
/usr/local/lib/python2.7/dist-packages/matplotlib/font_manager.py:273: UserWarning: Matplotlib is building the font cache using fc-list. This may take a moment.
  warnings.warn('Matplotlib is building the font cache using fc-list. This may take a moment.')
WARNING: unknown opcode starting with 36 32 5D 5B...)
WARNING: unknown opcode starting with 32 5D 5B 30...)
WARNING: unknown opcode starting with 5D 5B 30 5D...)
WARNING: unknown opcode starting with 5B 30 5D C0...)
WARNING: unknown opcode starting with 30 5D C0 06...)
WARNING: unknown opcode starting with 5D C0 06 00...)
WARNING: unknown opcode starting with C0 06 00 00...)
WARNING: unknown opcode starting with 06 00 00 5B...)
WARNING: unknown opcode starting with 5B 54 72 75...)
WARNING: unknown opcode starting with 54 72 75 65...)
WARNING: unknown opcode starting with 72 75 65 5D...)
WARNING: unknown opcode starting with 75 65 5D 00...)
WARNING: unknown opcode starting with 65 5D 00 1B...)
WARNING: unknown opcode starting with 5D 00 1B 69...)
WARNING: unknown opcode starting with 36 34 5D 1B...)
WARNING: unknown opcode starting with 34 5D 1B 69...)
WARNING: unknown opcode starting with 5D 1B 69 41...)
WARNING: unknown opcode starting with 31 5D 1B 69...)
WARNING: unknown opcode starting with 5D 1B 69 4B...)
WARNING: unknown opcode starting with 38 5D 1B 69...)
WARNING: unknown opcode starting with 5D 1B 69 64...)
WARNING: unknown opcode starting with 32 5D 67 00...)
WARNING: unknown opcode starting with 5D 67 00 5B...)
Traceback (most recent call last):
  File "/usr/local/bin/brother_ql_analyse", line 9, in <module>
    load_entry_point('brother-ql==0.6.dev0', 'console_scripts', 'brother_ql_analyse')()
  File "/usr/local/lib/python2.7/dist-packages/brother_ql/brother_ql_analyse.py", line 17, in main
    br.analyse()
  File "/usr/local/lib/python2.7/dist-packages/brother_ql/reader.py", line 238, in analyse
    for instruction in chunker(instructions):
  File "/usr/local/lib/python2.7/dist-packages/brother_ql/reader.py", line 138, in chunker
    num_bytes += data[2] + 2
TypeError: cannot concatenate 'str' and 'int' objects

PanderMusubi, 2016-09-20:
Which matplotlib packages are required to be installed?

A similar error occurs with:

# brother_ql_print test.bin  /dev/usb/lp0 
/usr/local/lib/python2.7/dist-packages/matplotlib/font_manager.py:273: UserWarning: Matplotlib is building the font cache using fc-list. This may take a moment.
  warnings.warn('Matplotlib is building the font cache using fc-list. This may take a moment.')

resulting in infintely red blinking light on QL-570 and no printing.

Trying other way to print also fails:

# cat test.bin> /dev/usb/lp0
cat: write error: Input/output error

Printing via USB address also fails:

# brother_ql_print  test.bin usb://0x04f9:0x2028
Traceback (most recent call last):
  File "/usr/local/bin/brother_ql_print", line 9, in <module>
    load_entry_point('brother-ql==0.6.dev0', 'console_scripts', 'brother_ql_print')()
  File "/usr/local/lib/python2.7/dist-packages/brother_ql/brother_ql_print.py", line 85, in main
    result = interpret_response(data)
  File "/usr/local/lib/python2.7/dist-packages/brother_ql/reader.py", line 150, in interpret_response
    assert len(data) >= 32
AssertionError

pklaus, 2016-09-20:
The error that happens from the initial issue description is rather a Python2 compat problem. The package was created for Python3. I didn't test it on Python2! The matplotlib warning is not a problem here, just a "warning". Your follow-up comment is about a different problem. Usually the printer's status word is 32 bytes long. What printer are you trying to use? Unfortunately I have to catch a flight and have to leave you with the issues for the next 14 days. Sorry.

PanderMusubi, 2016-09-20:
Ah, so switching over to Python 3 (which I prefer) would sove this? I need it working Thursday next week. :fearful: any help would be appreciated.

So the installation should be pip3 instead of pip?!

PanderMusubi, 2016-09-20:
With pip3 the analyse only gives warnings but provides proper output:

/usr/lib/python3/dist-packages/matplotlib/font_manager.py:273: UserWarning: Matplotlib is building the font cache using fc-list. This may take a moment.
  warnings.warn('Matplotlib is building the font cache using fc-list. This may take a moment.')
/usr/lib/python3/dist-packages/matplotlib/font_manager.py:273: UserWarning: Matplotlib is building the font cache using fc-list. This may take a moment.
  warnings.warn('Matplotlib is building the font cache using fc-list. This may take a moment.')
Page saved as page0001.png

Perhaps hide or prevent those warnings.

PanderMusubi, 2016-09-20:
Printing as root now is:

a) `cat test.bin > /dev/usb/lp0` results in no print and indefenite red blinking light until power cycle.

b) `brother_ql_print test.bin /dev/usb/lp0` results in no print and indefenite red blinking light until power cycle and error:

    ERROR:brother_ql.reader:Error: Transmission / Communication error
    ERROR:brother_ql.brother_ql_print:Errors occured: ['Transmission / Communication error']
    Traceback (most recent call last):
      File "/usr/local/bin/brother_ql_print", line 9, in <module>
        load_entry_point('brother-ql==0.6.dev0', 'console_scripts', 'brother_ql_print')()
      File "/usr/local/lib/python3.5/dist-packages/brother_ql/brother_ql_print.py", line 80, in main
        data = printer.read()
      File "/usr/local/lib/python3.5/dist-packages/brother_ql/backends/generic.py", line 36, in read
        ret_bytes = self._read(length)
      File "/usr/local/lib/python3.5/dist-packages/brother_ql/backends/linux_kernel.py", line 68, in _read
        data += os.read(self.read_dev, length)
    OSError: [Errno 19] No such device

c) `brother_ql_print test.bin usb://0x04f9:0x2028` results in no print and indefenite red blinking light until power cycle and error:

    ERROR:brother_ql.reader:Error: Transmission / Communication error
    ERROR:brother_ql.brother_ql_print:Errors occured: ['Transmission / Communication error']
    WARNING:brother_ql.brother_ql_print:Printing potentially not successful?

pklaus, 2016-09-20:
Which printer? Not all suppport all commands. It's in the Brother QL
programming guide... which I tried to reflect in the process of putting
together the commands. I successfully used the QL500 and QL710 printers.

Note that the brother_ql_create is currently hard-coded to create
instruction files for 62mm endless media. This was due to a lack of time on
my side so far. If you want, you can fix it. The place to change it is:
https://github.com/pklaus/brother_ql/blob/master/brother_ql/brother_ql_create.py#L80

PanderMusubi, 2016-09-20:
QL-570 with 62mm endless media. I can send the output of the debugger.

PanderMusubi, 2016-09-20:
This works:

    cat page0001.png | lpr -P Brother_QL-570
    inkscape --without-gui --export-pdf=/dev/stdout drawing.svg | lpr -P Brother_QL-
570

PanderMusubi, 2016-11-01:
PRiority is less now. However, when you have fix for above errors, I am willing to retest it all.

pklaus, 2016-11-20:
Hi @PanderMusubi, I think the problem with the QL-570 should be fixed now. Please refer to [#7](https://github.com/pklaus/brother_ql/issues/7#issuecomment-259640952). I'll close this issue here. If you still experience trouble with the QL-570, could you please consider reopening the other issue (because it's specific to that model) or opening a new one?
Thanks.
Also note that the restriction of `brother_ql_create` to 62mm endless labels was removed. Now, the tool accepts a `--label-size` parameter to work with (hopefully) any label offered by Brother.

closed 2016-11-20 by pklaus