pklaus/brother_ql #7: [Brother QL-570] Red flashing while printing

enagra, 2016-10-08:
I'm using QL-570 Model. Label type: [Prestige Cartridge RB22205](https://www.amazon.es/gp/product/B00F43SHEO/ref=oh_aui_detailpage_o07_s00?ie=UTF8&psc=1) (62 mm x 30.48 m).

**Printing:**
```
brother_ql_print Desktop/etiqueta.bin usb://0x04f9:0x2028
ERROR:brother_ql.reader:Error: Transmission / Communication error
ERROR:brother_ql.brother_ql_print:Errors occured: ['Transmission / Communication error']
WARNING:brother_ql.brother_ql_print:Printing potentially not successful?
[1]+  Terminated: 15          brother_ql_print Desktop/etiqueta.bin usb://0x04f9:0x2028
```

**Red flashing while printing and Utility status monitor printer:**
```
QL-570
USB
62mm
Ha ocurrido un error por una causa desconocida (An error has occurred from an unknown cause)
```

**Debugging:**
```
brother_ql_debug Desktop/etiqueta.bin usb://0x04f9:0x2028
INFO: CMD preamble FOUND. Instruction: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [...] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
INFO: CMD init FOUND. Instruction: 1B 40 
INFO: CMD status request FOUND. Instruction: 1B 69 53 
INFO: Response from the device: 80 20 42 34 32 30 00 00 00 00 3E 0A 00 00 15 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
INFO: Interpretation of the response: 'Reply to status request' (phase: Waiting to receive), 'Continuous length tape' 62x0 mm^2, errors: []
INFO: CMD media/quality FOUND. Instruction: 1B 69 7A CE 0A 3E 00 0A 01 00 00 01 00 
INFO: CMD various FOUND. Instruction: 1B 69 4D 40 
INFO: CMD cut-every FOUND. Instruction: 1B 69 41 01 
INFO: CMD expanded FOUND. Instruction: 1B 69 4B 08 
INFO: CMD margins FOUND. Instruction: 1B 69 64 23 00 
INFO: CMD compression FOUND. Instruction: 4D 02 
INFO: CMD raster FOUND. Instruction: 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 [...] 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 
INFO: Response from the device: 80 20 42 34 32 30 00 00 00 04 3E 0A 00 00 15 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00
ERROR: Error: Transmission / Communication error
INFO: Interpretation of the response: 'Error occurred' (phase: Waiting to receive), 'Continuous length tape' 62x0 mm^2, errors: ['Transmission / Communication error']
INFO: CMD print FOUND. Instruction: 1A
```

**Testing with your file:**
```
brother_ql_print Desktop/720x151_monochrome.png usb://0x04f9:0x2028
ERROR:brother_ql.reader:Error: Transmission / Communication error
ERROR:brother_ql.brother_ql_print:Errors occured: ['Transmission / Communication error']
WARNING:brother_ql.brother_ql_print:Printing potentially not successful?
```

**More files attached:** (brother_ql_analyse and images)
[Archivo comprimido.zip](https://github.com/pklaus/brother_ql/files/517629/Archivo.comprimido.zip)

pklaus, 2016-11-10:
It looks like I was allowing the TIFF compression mode for the QL-570 in my code but the device doesn't support it. Can you please upgrade the package, create the raster instruction file again and try to print again? Thanks!

closed 2016-11-20 by pklaus

enagra, 2016-12-29:
Thanks for your answer. It is very difficult for me to have enough time (two babies) to sit, try it out and try to publish correctly. I had this project paralyzed. I'm sorry I took so long to respond. Next time I'll do it sooner.

I have the same report, with the same result after the updates. I made a mistake: (calling brother_ql_print with a PNG file). It was an oversight.

Can you give me an image so I can try it? In this way, it would rule out a formatting problem. I do not know what else I can do to help ...

Thank you for your work. Here is the last test of the last update:

**Udating last version**
```
pip install --upgrade https://github.com/pklaus/brother_ql/archive/master.zip

Collecting https://github.com/pklaus/brother_ql/archive/master.zip
  Downloading https://github.com/pklaus/brother_ql/archive/master.zip
Requirement already up-to-date: future in /Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages (from brother-ql==0.7.dev0)
Requirement already up-to-date: packbits in /Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages (from brother-ql==0.7.dev0)
Requirement already up-to-date: pillow in /Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages (from brother-ql==0.7.dev0)
Installing collected packages: brother-ql
  Found existing installation: brother-ql 0.6.dev0
    Uninstalling brother-ql-0.6.dev0:
      Successfully uninstalled brother-ql-0.6.dev0
  Running setup.py install for brother-ql ... done
Successfully installed brother-ql-0.7.dev0
You are using pip version 8.1.2, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

**Printing:**

```
brother_ql_print Desktop/etiqueta.bin usb://0x04f9:0x2028
ERROR:brother_ql.reader:Error: Transmission / Communication error
ERROR:brother_ql.brother_ql_print:Errors occured: ['Transmission / Communication error']
WARNING:brother_ql.brother_ql_print:Printing potentially not successful?
```

**Red flashing while printing and Utility status monitor printer:**

```
QL-570
USB
62mm
Ha ocurrido un error por una causa desconocida (An error has occurred from an unknown cause)
```

**Debugging:**

```
brother_ql_debug Desktop/etiqueta.bin usb://0x04f9:0x2028
INFO: CMD preamble FOUND. Instruction: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [...] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
INFO: CMD init FOUND. Instruction: 1B 40 
INFO: CMD status request FOUND. Instruction: 1B 69 53 
INFO: CMD media/quality FOUND. Instruction: 1B 69 7A CE 0A 3E 00 0A 01 00 00 01 00 
INFO: Response from the device: 80 20 42 34 32 30 00 00 00 00 3E 0A 00 00 15 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
INFO: Interpretation of the response: 'Reply to status request' (phase: Waiting to receive), 'Continuous length tape' 62x0 mm^2, errors: []
INFO: CMD various FOUND. Instruction: 1B 69 4D 40 
INFO: CMD cut-every FOUND. Instruction: 1B 69 41 01 
INFO: CMD expanded FOUND. Instruction: 1B 69 4B 08 
INFO: CMD margins FOUND. Instruction: 1B 69 64 23 00 
INFO: CMD compression FOUND. Instruction: 4D 02 
INFO: CMD raster FOUND. Instruction: 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 [...] 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 67 00 02 A7 00 
INFO: Response from the device: 80 20 42 34 32 30 00 00 00 04 3E 0A 00 00 15 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00
ERROR: Error: Transmission / Communication error
INFO: Interpretation of the response: 'Error occurred' (phase: Waiting to receive), 'Continuous length tape' 62x0 mm^2, errors: ['Transmission / Communication error']
INFO: CMD print FOUND. Instruction: 1A
```

pklaus, 2016-12-29:
No problem for the delay :)
Well, it looks like you were printing again, but **not** generating the .bin file again in the first place. I can see this from the `INFO: CMD compression FOUND. Instruction: 4D 02` which should not be in there since [13a66a8](https://github.com/pklaus/brother_ql/commit/13a66a8b1be8d7ae7a7d5a6606d3dec1165faa39). Can you please check that again. Thanks.

enagra, 2016-12-29:
Great! I just tried it and it works fine. I did not realize I had to regenerate the BIN.

Thanks for your time. Have a good new year!

pklaus, 2016-12-29:
Great! All the best for you and your family in 2017!