pklaus/brother_ql #8: [Brother QL-570] Black Square Printout

nmaas87, 2016-11-22:
Hello pklaus and thanks a lot for the awesome idea!
I am using QL-570 with Raspberry Pis as Invoice Printer, however, the cups drivers are less than usable and so I stumbled upon your tool - and really want to use it (as my input data is transfered as png files via python.. so.. could be good 👍)

However, the setup sadly does not work.
I am using an QL-570 with  54 / 590 / (54 mm endless) tape and the -to-be-wished- result would be an all page print out of the reciepe as seen in the added file. However, I only get an black square with the first file (cafe.png) and an black square with less "white space" / tape (cafe90.png). I included the bin files and png files as well.

And following errors:

`root@demo:/home/pi/test# brother_ql_create --model QL-570 --label-size 54 ./cafe.png cafe.bin
root@demo:/home/pi/test# brother_ql_create --model QL-570 --label-size 54 ./cafe90.png cafe90.bin
root@demo:/home/pi/test# brother_ql_print cafe.bin /dev/usb/lp0
Traceback (most recent call last):
  File "/usr/local/bin/brother_ql_print", line 9, in <module>
    load_entry_point('brother-ql==0.6-dev', 'console_scripts', 'brother_ql_print')()
  File "/usr/local/lib/python2.7/dist-packages/brother_ql/brother_ql_print.py", line 85, in main
    result = interpret_response(data)
  File "/usr/local/lib/python2.7/dist-packages/brother_ql/reader.py", line 157, in interpret_response
    if error_info_1 & (1 << error_bit):
TypeError: unsupported operand type(s) for &: 'str' and 'int'
root@demo:/home/pi/test# brother_ql_print cafe90.bin /dev/usb/lp0
Traceback (most recent call last):
  File "/usr/local/bin/brother_ql_print", line 9, in <module>
    load_entry_point('brother-ql==0.6-dev', 'console_scripts', 'brother_ql_print')()
  File "/usr/local/lib/python2.7/dist-packages/brother_ql/brother_ql_print.py", line 85, in main
    result = interpret_response(data)
  File "/usr/local/lib/python2.7/dist-packages/brother_ql/reader.py", line 157, in interpret_response
    if error_info_1 & (1 << error_bit):
TypeError: unsupported operand type(s) for &: 'str' and 'int'
`

You can download the file here https://jctixx.de/download/QL570Test.zip

If I can help you out with test prints or such, just let me know.
Also, I tried to play with the treshold setting - it did not make it less or more black :(

pklaus, 2016-11-22:
Thanks for the bug report. This issue should be fixed in 743d963. Please upgrade and test again.

nmaas87, 2016-11-22:
Absolutely awesome! It works like a treat! 👍 Thanks a lot for the very quick help!
Do you have any intention to make that program into a kind of  Python API, so that you could call it as a function with an file and the needed variables to include that in own software? (It ain't needed, as one can really fastly call it via shell commands and such, but just as a question :)!)

pklaus, 2016-11-22:
By the way: The errors in brother_ql_print show that you are using it with Python 2. It was not made for this although I spent some work to make brother_ql_create work with Python 2. You can simply print by using `cat cafe.bin > /dev/usb/lp0` instead of `brother_ql_print`.

pklaus, 2016-11-22:
You can do:

    from brother_ql import BrotherQLRaster, create_label
    qlr = BrotherQLRaster("QL-570")
    # image can be a PIL/Pillow Image() object or a file name to the GIF/PNG/JPEG image:
    create_label(qlr, image, "62", threshold=70, cut=True)
    # The binary image output is then available in this variable:
    qlr.data

You can send this to your printer via simple Python:

    import os
    dev = os.open('/dev/usb/lp0', os.O_RDWR)
    try:
        data = os.write(dev, qlr.data)
    except KeyboardInterrupt:
        pass
    finally:
        os.close(dev)

Or on Python 3 you can use brother_ql's more high-level variants:

    from brother_ql.backends.linux_kernel import BrotherQLBackendLinuxKernel as BrotherQLBackend
    printer = BrotherQLBackend("/dev/usb/lp0")
    printer.write(qlr.data)

closed 2016-11-22 by pklaus