pklaus/brother_ql #10: raspberry pi

axxg, 2016-12-02:
Hi,
nice script, but i have a problem at a raspberry pi 3 with raspbian:
the log:
copying lib/matplotlib/tests/baseline_images/test_axes/pcolormesh.png -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_axes

copying lib/matplotlib/tests/baseline_images/test_axes/test_eventplot_problem_kwargs.png -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_axes

copying lib/matplotlib/tests/baseline_images/test_axes/step_linestyle.pdf -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_axes

copying lib/matplotlib/tests/baseline_images/test_axes/formatter_ticker_004.pdf -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_axes

copying lib/matplotlib/tests/baseline_images/test_axes/contour_colorbar.svg -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_axes

copying lib/matplotlib/tests/baseline_images/test_axes/errorbar_zorder.pdf -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_axes

copying lib/matplotlib/tests/baseline_images/test_axes/log_scales.svg -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_axes

copying lib/matplotlib/tests/baseline_images/test_axes/single_date.svg -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_axes

copying lib/matplotlib/tests/baseline_images/test_axes/polar_wrap_360.pdf -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_axes

copying lib/matplotlib/tests/baseline_images/test_axes/bxp_withmean_point.png -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_axes

copying lib/matplotlib/tests/baseline_images/test_axes/pie_frame_grid.png -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_axes

copying lib/matplotlib/tests/baseline_images/test_axes/hist_stacked.png -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_axes

copying lib/matplotlib/tests/baseline_images/test_axes/units_strings.svg -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_axes

copying lib/matplotlib/tests/baseline_images/test_axes/nonfinite_limits.pdf -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_axes

copying lib/matplotlib/tests/baseline_images/test_axes/violinplot_horiz_showall.png -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_axes

copying lib/matplotlib/tests/baseline_images/test_axes/hist_log.png -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_axes

copying lib/matplotlib/mpl-data/fonts/ttf/LICENSE_STIX -> build/lib.linux-armv7l-2.7/matplotlib/mpl-data/fonts/ttf

creating build/lib.linux-armv7l-2.7/matplotlib/backends/web_backend/jquery/js

copying lib/matplotlib/backends/web_backend/jquery/js/jquery-1.11.3.min.js -> build/lib.linux-armv7l-2.7/matplotlib/backends/web_backend/jquery/js

copying lib/matplotlib/backends/web_backend/jquery/js/jquery-ui.min.js -> build/lib.linux-armv7l-2.7/matplotlib/backends/web_backend/jquery/js

copying lib/matplotlib/mpl-data/fonts/ttf/README.TXT -> build/lib.linux-armv7l-2.7/matplotlib/mpl-data/fonts/ttf

creating build/lib.linux-armv7l-2.7/matplotlib/backends/web_backend/css

copying lib/matplotlib/backends/web_backend/css/page.css -> build/lib.linux-armv7l-2.7/matplotlib/backends/web_backend/css

copying lib/matplotlib/backends/web_backend/css/fbm.css -> build/lib.linux-armv7l-2.7/matplotlib/backends/web_backend/css

copying lib/matplotlib/backends/web_backend/css/boilerplate.css -> build/lib.linux-armv7l-2.7/matplotlib/backends/web_backend/css

creating build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_spines

copying lib/matplotlib/tests/baseline_images/test_spines/spines_capstyle.pdf -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_spines

copying lib/matplotlib/tests/baseline_images/test_spines/spines_data_positions.png -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_spines

copying lib/matplotlib/tests/baseline_images/test_spines/spines_axes_positions.pdf -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_spines

copying lib/matplotlib/tests/baseline_images/test_spines/spines_axes_positions.svg -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_spines

copying lib/matplotlib/tests/baseline_images/test_spines/spines_capstyle.svg -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_spines

copying lib/matplotlib/tests/baseline_images/test_spines/spines_data_positions.svg -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_spines

copying lib/matplotlib/tests/baseline_images/test_spines/spines_axes_positions.png -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_spines

copying lib/matplotlib/tests/baseline_images/test_spines/spines_data_positions.pdf -> build/lib.linux-armv7l-2.7/matplotlib/tests/baseline_images/test_spines
[... 521 lines cut ...]
building 'matplotlib.ft2font' extension

creating build/temp.linux-armv7l-2.7

creating build/temp.linux-armv7l-2.7/src

arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DPY_ARRAY_UNIQUE_SYMBOL=MPL_matplotlib_ft2font_ARRAY_API -DNPY_NO_DEPRECATED_API=NPY_1_7_API_VERSION -I/usr/lib/python2.7/dist-packages/numpy/core/include -I/usr/include/freetype2 -I/usr/local/include -I/usr/include -I. -I/usr/include/python2.7 -c src/ft2font.cpp -o build/temp.linux-armv7l-2.7/src/ft2font.o

In file included from src/ft2font.cpp:9:0:

src/mplutils.h:31:20: fatal error: Python.h: Datei oder Verzeichnis nicht gefunden

 #include <Python.h>

                    ^

compilation terminated.

error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

----------------------------------------
Cleaning up...
Command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-Z9mhMv/matplotlib/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-D6YooL-record/install-record.txt --single-version-externally-managed --compile failed with error code 1 in /tmp/pip-build-Z9mhMv/matplotlib
Storing debug log for failure in /root/.pip/pip.log
pi@raspberrypi:~/Downloads $

pklaus, 2016-12-10:
Hi @axxg, thanks for the report. (I allowed myself to shorten the error output to the relevant lines.)

**In short: This is fixed in the commit c18c3f6 by removing the dependency on the package matplotlib.**

Details:
The problem is that you are missing the python-dev package on your Raspberry Pi. This means that the installation of the matplotlib package (a dependency of this software at the time) was failing. Actually, I don't use this package anymore in brother_ql and thus I kicked out that requirement. I should have done that earlier. I also removed another nasty dependency, numpy in the commit 17274a8. It's a pretty standard package but the installation from sources (done by the `pip install ...` command) takes a while on the Raspberry Pi and I was not using the package heavily.

So thanks for reporting. And please check again, if it works for you now.

pklaus, 2016-12-15:
I'm closing this now. Let me know if you are still having problems.

closed 2016-12-15 by pklaus