pklaus/brother_ql #11: Printer not finishing job correctly

robincalmegard, 2016-12-11:
Hi there, thank your for your awesome work! I am building stuff with a Raspberry Pi and a Brother QL-570 printer.

However, I have a problem when printing the generated bin. After printing, the printer won't cut the label off - and it also becomes unresponsive (not accepting any new print jobs) until I restart it with the power button. Seems like something isn't finished correctly.

Debug:

```
pi@raspberrypi:~ $ brother_ql_create --model QL-570 --label-size 29x90 --loglevel DEBUG ./image.png > label.bin
INFO:brother_ql.raster:raster_image_size: 720x991

pi@raspberrypi:~ $ brother_ql_print --debug label.bin /dev/usb/lp0
INFO:brother_ql.brother_ql_print:Sending instructions to the printer. Total: 92306 bytes.
DEBUG:brother_ql.backends.generic:Writing 92306 bytes.
DEBUG:brother_ql.backends.generic:Read 32 bytes.
DEBUG:brother_ql.reader:Byte  0         Print head mark: 80
DEBUG:brother_ql.reader:Byte  1                    Size: 20
DEBUG:brother_ql.reader:Byte  2          Fixed (B=0x42): 42
DEBUG:brother_ql.reader:Byte  3        Device dependent: 34
DEBUG:brother_ql.reader:Byte  4        Device dependent: 32
DEBUG:brother_ql.reader:Byte  5          Fixed (0=0x30): 30
DEBUG:brother_ql.reader:Byte  6  Fixed (0x00 or 0=0x30): 00
DEBUG:brother_ql.reader:Byte  7            Fixed (0x00): 00
DEBUG:brother_ql.reader:Byte  8     Error information 1: 00
DEBUG:brother_ql.reader:Byte  9     Error information 2: 00
DEBUG:brother_ql.reader:Byte 10             Media width: 1D
DEBUG:brother_ql.reader:Byte 11              Media type: 0B
DEBUG:brother_ql.reader:Byte 12            Fixed (0x00): 00
DEBUG:brother_ql.reader:Byte 13            Fixed (0x00): 00
DEBUG:brother_ql.reader:Byte 14                Reserved: 01
DEBUG:brother_ql.reader:Byte 15                    Mode: 00
DEBUG:brother_ql.reader:Byte 16            Fixed (0x00): 00
DEBUG:brother_ql.reader:Byte 17            Media length: 5A
DEBUG:brother_ql.reader:Byte 18             Status type: 00
DEBUG:brother_ql.reader:Byte 19              Phase type: 00
DEBUG:brother_ql.reader:Byte 20     Phase number (high): 00
DEBUG:brother_ql.reader:Byte 21      Phase number (low): 00
DEBUG:brother_ql.reader:Byte 22     Notification number: 00
DEBUG:brother_ql.reader:Byte 23                Reserved: 00
DEBUG:brother_ql.reader:Byte 24                Reserved: 00
DEBUG:brother_ql.reader:Media type: Die-cut labels
DEBUG:brother_ql.reader:Status type: Reply to status request
DEBUG:brother_ql.reader:Phase type: Waiting to receive
DEBUG:brother_ql.brother_ql_print:TIME 3.172 - result: {'status_type': 'Reply to status request', 'errors': [], 'media_type': 'Die-cut labels', 'phase_type': 'Waiting to receive', 'media_width': 29, 'media_length': 90}
DEBUG:brother_ql.backends.generic:Read 32 bytes.
DEBUG:brother_ql.reader:Byte  0         Print head mark: 80
DEBUG:brother_ql.reader:Byte  1                    Size: 20
DEBUG:brother_ql.reader:Byte  2          Fixed (B=0x42): 42
DEBUG:brother_ql.reader:Byte  3        Device dependent: 34
DEBUG:brother_ql.reader:Byte  4        Device dependent: 32
DEBUG:brother_ql.reader:Byte  5          Fixed (0=0x30): 30
DEBUG:brother_ql.reader:Byte  6  Fixed (0x00 or 0=0x30): 00
DEBUG:brother_ql.reader:Byte  7            Fixed (0x00): 00
DEBUG:brother_ql.reader:Byte  8     Error information 1: 00
DEBUG:brother_ql.reader:Byte  9     Error information 2: 00
DEBUG:brother_ql.reader:Byte 10             Media width: 1D
DEBUG:brother_ql.reader:Byte 11              Media type: 0B
DEBUG:brother_ql.reader:Byte 12            Fixed (0x00): 00
DEBUG:brother_ql.reader:Byte 13            Fixed (0x00): 00
DEBUG:brother_ql.reader:Byte 14                Reserved: 01
DEBUG:brother_ql.reader:Byte 15                    Mode: 00
DEBUG:brother_ql.reader:Byte 16            Fixed (0x00): 00
DEBUG:brother_ql.reader:Byte 17            Media length: 5A
DEBUG:brother_ql.reader:Byte 18             Status type: 06
DEBUG:brother_ql.reader:Byte 19              Phase type: 01
DEBUG:brother_ql.reader:Byte 20     Phase number (high): 00
DEBUG:brother_ql.reader:Byte 21      Phase number (low): 00
DEBUG:brother_ql.reader:Byte 22     Notification number: 00
DEBUG:brother_ql.reader:Byte 23                Reserved: 00
DEBUG:brother_ql.reader:Byte 24                Reserved: 00
DEBUG:brother_ql.reader:Media type: Die-cut labels
DEBUG:brother_ql.reader:Status type: Phase change
DEBUG:brother_ql.reader:Phase type: Printing state
DEBUG:brother_ql.brother_ql_print:TIME 3.178 - result: {'status_type': 'Phase change', 'errors': [], 'media_type': 'Die-cut labels', 'phase_type': 'Printing state', 'media_width': 29, 'media_length': 90}
```

After this, it takes a few seconds until the following shows up:

`WARNING:brother_ql.brother_ql_print:Printing potentially not successful?`

Any ideas on what might be causing this? Model number and label size is correct.

Thanks.

pklaus, 2016-12-12:
Thanks for the report. I also have some problems with printouts on my QL-500 currently and I will fix this within the next days hopefully. Just for curiosity: does the problem also occur, if you print with `cat label.bin > /dev/usb/lp0` ?

robincalmegard, 2016-12-12:
Great, thanks for looking into it! Yes, it happens when I print it directly to the device aswell.

nmaas87, 2016-12-12:
I tried and got the same problem with the QL-570, just to affirm :)

pklaus, 2016-12-15:
I commited a change that fixes the problem on my QL-500. Could you please try this new version? (You can upgrade brother_ql to the latest version here on Github by entering: `pip install --upgrade --no-deps https://github.com/pklaus/brother_ql/archive/master.zip`).

nmaas87, 2016-12-15:
I'll do so right away, it will take some time, but I will write you as soon as I got it finished (1-2 hours) (on QL-570) :)

robincalmegard, 2016-12-15:
I don't have the QL-570 anymore, had a deadline and needed a quick solution. But will definitely check back when I need to do this again - which may be quite soon. :) Good luck with the project!

nmaas87, 2016-12-15:
This issue can be closed, I can confirm that it is working with a QL-570 as intended, again.
I cancled my build process as soon as I saw commit 44228e6b16459c607db3be21c58ea80474ea60e4 comin' up and did restart it, so I kind of tested the latest version in the repo and it worked great. Thanks a lot!