pklaus/brother_ql #15: Print speed

iegomez, 2017-02-10:
I was using your package on an Aria G25 by Acme Systems (it's packed with a 400 MHz Atmel processor) to print to a QL-710W and it worked fine, the problem was speed: it took about 8 seconds to convert the png to raster and then print it.
I did some digging and found a pair of bottlenecks. The first one is the frame b/w invertion on the raster's add_raster_data function. That alone took around 5 seconds. The second one was the string concatenation of the raster data in the same function, which took around 2 seconds. So I made some workarounds and got the whole process to just over 1 second.

First, instead of doing your own inversion, you could lest Pillow invert the png just after resize:

`im = PIL.ImageOps.invert(im)`

Then, avoid string concatenation overhead using whatever better method you prefer. I did it by writing to a pseudo file and then setting the data to the file's value:

```
        file_str = StringIO()

        while start + row_len <= frame_len:
            row = frame[start:start+row_len]
            start += row_len
            file_str.write(b'\x67\x00')
            if self._compression:
                row = packbits.encode(row)
            file_str.write(bytes([len(row)]))
            file_str.write(row)

        self.data = file_str.getvalue()
```

There's one more optimization you may use in that loop if your device has more than once core (mine doesn't): you could parallelize the frame to data transfer. For example, if your device has 2 cores, just start 2 threads/processes and let each one deal with half the frame, then write to data in order. In my Aria of course that didn't help, but you should see an improvement in a Rasperry Pi, for example.

All this was tested with a CPython implementaion of Python 2.7. Anyway, just thought I should mention it if anyone is trying to use your package in a low-end device. If you like my proposals, I could clone your repo, make the changes and do a pull request, or you could just implement them yourself if you prefer. Let me know.

Cheers.

dforrer, 2017-02-20:
I would be interested in trying out the changes you have made. 
I am doing a project on the Raspberry Pi 3 and the print speed is a bottleneck.

Could you make a fork so that we can see your exact changes?

iegomez, 2017-02-20:
Hi, I'll try to fork it and add the changes this week. I further modified the package for some custom needs, so I'll have to make another working copy and try it. I'll let you know in these comments when I'm done so you can give it a try.

iegomez, 2017-02-20:
Just a detail: do you intend to call the package from a terminal or other program, or are you including it on a python program? Is just to know if you'll need a command line option to set how many cores to use, or just being able to call the function directly with the number of cores is fine to start with.

pklaus, 2017-02-20:
Hi there,
first of all: thanks a lot @iegomez for doing the hard work searching for the bottleneck. I was working on the package with a fast Desktop computer and a Raspberry Pi 3 otherwise so I didn't feel the same pressure to search hard. I gladly incorporate it into the package but this week I'm busy with my everyday job. If you fork the package and submit a pull request, I'll definitely review and merge it! I'm also happy to accept a new command line parameter. I would let it default to the number of cores the machine has.

iegomez, 2017-02-20:
Hi, @pklaus.
Thank you for creating this package, it really saved my day a couple of weeks ago.
I managed to get some time today and implemented the changes, including an extra command line parameter. I also took your suggestion and defaulted to the number of cores of the machine. You can pass that param with the --cores (or -c) option.

There's one detail with the implementation: the image inverting is done earlier now, so the threshold gets reversed (when you used 70 before, now you need a threshold of 30 to achieve the same result). Also, it depends on PIL.ImageOps, which I had to import. Other further imports are those concerning multiprocessing that I added to the raster file.

I've done the pull request with this same message so you can review it.

Cheers.

pklaus, 2017-09-18:
Thanks for your effort to report this issue / your findings. As the continuation of this discussion was anyhow ongoing in pull request #16 I will close this issue. If there are further aspects, feel free to add it to the pull request discussion (and ignore that it's "closed" as well).

closed 2017-09-18 by pklaus