pklaus/brother_ql #18: 600x300 dpi support

raidoz, 2017-06-11:
What needs to be done to support the 600x300 dpi mode? I have a QL-710W and am willing to help.

pklaus, 2017-06-17:
Hi! After all, the high-dpi mode is just a bit of difference [here](https://github.com/pklaus/brother_ql/blob/master/brother_ql/brother_ql_create.py#L137) and [here](https://github.com/pklaus/brother_ql/blob/master/brother_ql/raster.py#L149).
More work has to be spent converting your images to 600x300dpi. Maybe you would work with 600x600 and the do the summing across the columns manually?
Why do you feel you need this, 300x300dpi are not enough? What's your application? Barcode printing?

fmulp0, 2017-10-01:
Hi! I'm also interested in the 600dpi printing because I didn't find any information how to print in high dpi mode using ESC/P and I need it because my barcodes aren't readable with my barcode scanner when printed in 300dpi.

pklaus, 2017-10-01:
Thanks for your feedback, @fmulp0.
Can you please provide the picture you want to print with 600 dpi? Maybe via email so that the quality doesn't suffer from the upload to Github.
I think a new CLI flag `--600dpi` would be the way to go. The picture should be 600 dpi in both dimensions and *brother_ql* than downscales one dimension with a divisor of 2 so that we get a 600x300 dpi image (required for this to work).

pklaus, 2017-10-01:
Commit a354cde now enables 600dpi printing. Please try it out (`pip install --upgrade https://github.com/pklaus/brother_ql/archive/master.zip`) and report any problems.

closed 2017-10-06 by pklaus