pklaus/brother_ql #19: QL-800 series support

raidoz, 2017-06-11:
How likely is it that the QL-800 series works by just adding the relevant entries to devicedependent.py? It also has the ability to somehow print red on some labels - that probably needs extra work? I don't have the printer yet, but I need a printer for a project and I would get the 800 to test, if it is likely to work with brother_ql.

pklaus, 2017-06-17:
Thanks for pointing out the fact that a new generation of printers is out. From reading in the raster language documentation I see no reason why it shouldn't work. The model would have to be added in `devicedependent.py` but you could try using another new model like QL-700 for generating the commands, first.
For printing in red on DK-22251 labels, a new command needs to be supported. This will only be the case if someone provides such a printer to me (preferably the 810/820) or if someone implements it and sends a pull request.

pklaus, 2017-09-16:
Good news: I was being provided a QL-820NWB (I'm grateful to the donor). Support is now underway. The commit 555a51a adds the basic device dependent features to the library. Commit f2c7dbc added support for the QL-800 series and their new commands to the `brother_ql_analyze` CLI tool.
It might take some more days until printing red/black/white on DK-22251 is supported.

pklaus, 2017-10-06:
Version v0.8.1 should now fully support printing on black/red/white labels (DK-22251).

closed 2017-10-06 by pklaus