pklaus/brother_ql #21: Cannot print 24 dia labels

mark77-DE, 2017-09-08:
While trying to create 24mm diameter labels, an error about mismatching picture dimentions occurs.

Interesting is, when you add:
`
logger.warning(im.size[0])
logger.warning(im.size[1])
logger.warning(dots_printable[0])
logger.warning(dots_printable[1])`
to "brother_ql_create.py", it will output 236 for each warning.

Anyhow, it will exit on:
`if im.size[0] != dots_printable[0] or im.size[1] != dots_printable[1]:
sys.exit("Check your image dimensions. Expecting: " + str(dots_printable))`

pklaus, 2017-09-17:
Thanks for reporting an issue.

I changed the reporting in the `sys.exit()` line in commit ee69e04. Could you please upgrade the package to the latest version on Github using:

    pip install --upgrade https://github.com/pklaus/brother_ql/archive/master.zip

And check again what the output is? Also, can you please add the image (or another example image of the same size) to this issue?

mark77-DE, 2017-09-18:
Thats working now, thanks a lot!

[image]

closed 2017-09-18 by mark77-DE