pklaus/brother_ql #23: printing with network backend fails

splitbrain, 2017-10-06:
I'm trying to print to a QL-500 printer attached to my router (which provides the print server) to a 62mm endless roll. When use `netcat` to send the printfile to the printer it works just fine. However when I use `brother_ql_print` only a the first line gets printed and then the printer needs to be resetted.

Debug output:

```
$ ./env/bin/brother_ql_print --debug --backend network test.bin tcp://192.168.1.1:9100
WARNING:brother_ql.brother_ql_print:The network backend doesn't supply any 'readback' functionality. No status reports will be received.
INFO:brother_ql.brother_ql_print:Sending instructions to the printer. Total: 290849 bytes.
DEBUG:brother_ql.backends.generic:Writing 290849 bytes.
```

The reported file size seems to be correct:

```
$ ls -l test.bin 
-rw-r--r-- 1 andi users 290849 Oct  6 11:05 test.bin
```
I attached the print file and the original png here. [test.zip](https://github.com/pklaus/brother_ql/files/1362187/test.zip)

Is there anything I can to do help debugging this? From what I understand the network backend just opens a socket and then writes the whole content to it. Could it be that there is some buffering involved that needs to be flushed? (I'm not a python developer).

splitbrain, 2017-10-06:
According to `socket.send()` documentation:

> Send a data string to the socket.  [...]  Return the number of bytes sent; this may be less than len(data) if the network is busy.

So I tried changing the function to this:

```python
    def _write(self, data):
        length = len(data)
        totalsent = 0
        while totalsent < length:
            sent = self.s.send(data[totalsent:])
            if sent == 0:
                raise RuntimeError("socket connection broken")
            totalsent = totalsent + sent
```

This is make the printer more of the file but still not all. The app crashes with a `socket.timeout: timed out` exception.

splitbrain, 2017-10-06:
Okay, so the timeout seems to define how long a whole send operation is allowed to take. The default of 0.01 is much too small. Using the following now prints successfully:

```python
    def _write(self, data):
        self.s.settimeout(10)
        self.s.sendall(data)
```

This probably now messes with read timeouts, so it might need to be coded a bit differently, but I guess you know better where and how, so I'm not opening a PR for this.

pklaus, 2017-10-06:
Dear @splitbrain, thanks for you report. I've never used any of the QL series printers with a print server. So it's a good use case to test against.

Concerning the timeout: the documentation of socket.sendall() [says](https://docs.python.org/3/library/socket.html#socket.socket.sendall) the behaviour with respect to `timeout` changed with Python 3.5:

> Changed in version 3.5: The socket timeout is no more reset each time data is sent successfully. The socket timeout is now the maximum total duration to send all data.

I think we should raise the timeout to a higher value as you suggest, to block in case of back-pressure on the socket.

pklaus, 2017-10-18:
Thanks again for reporting this. I finally pushed the fix and released version v0.8.2 which contains it. The bug could occur with either large labels or slow connections. It could also be triggered by not choosing compression (on the QL-710W for example) as this also results in a much larger instruction file. So thanks for spotting this problem. Please let me know if there are still problems.

closed 2017-10-18 by pklaus