pklaus/brother_ql #24: Is there a way to print multiple "pages"

caish5, 2017-10-12:
I saw in the ReadMe that it's possible to have multiple pages in the bin file. Any chance you could point me in the right direction as to how that is done?

pklaus, 2017-10-12:
Well, after all you can simply generate multiple .bin files and then concatenate them together.
The question is also what exactly you want to achieve. Do you have a multi page document format (PDF) that you want to print? Or do you want to generate smaller .bin files? Or do you want to print multiple die-cut lables and only cut the tape after the last one?
The latter can be achieved with the --no-cut option of brother_ql_create.

caish5, 2017-10-13:
I'm printing multiple copies of die cut labels with Raspberry Pis. (Use by dates in a restaurant). 
At present I just use the Brother drivers through qemu.
The downside of that is that it takes about 5-10 seconds to start printing. Once started however, the labels fly out of the printer at speed.
brother_ql starts printing very quickly. 
I tried concatenating as you suggested. It works but the printer realigns itself after each label which likely makes the total print time longer. Swings and Roundabouts I guess!

kevinsimper, 2018-06-19:
@caish5 How did you end up solving the issue?

I am thinking about batching writes, as starting to print takes a bit of time.

caish5, 2018-06-21:
I ended up continuing to use the Brother Drivers. I have recently discovered the open source printer-driver-ptouch which supports the QL-570 needs no qemu and is fast. It only included a limited set of label sizes in the ppd file though. I added the 23x23mm that I need manually from the brother PPD. But then the labels are offset too far too the right. 
So Close to perfect! But I'm missing something about then PPD because on the labels it does include, you get a perfect test page.

stuartlynne, 2018-09-02:
I have the same problem.

I am trying to print a PDF containing two 62x100 labels to QL-710W printers and need to have the cut after the second label. 

If I convert the PDF to two png files (ghostscript and brother_ql_create) and then convert those one at a time to raster bin files, the first with no-cut and the second with cut, the printer prints the first and second and cuts, then gets into a loop pushing blank labels out and cutting them about a cm short of the cut line (for die cut labels.) until the printer is turned off.

I can't figure out the correct option settings to convince brother_ql print to set options that convince the ql710w to not cut after every label. 

The exact same PDF run through ghostscript and piped through the brother rastertobrpt1 works correctly (two labels, cut after second), but of course the imaging time is far worse (3-4 seconds compared to .1 seconds for brother_ql).

stuartlynne, 2018-09-02:
> The latter can be achieved with the --no-cut option of brother_ql_create.

You can't pass multiple png files to brother_ql_create. If two files are specified it uses the second as the outfile. 

Again, what I'm trying to do is convert a multiple label pdf file to a single raster file that will printer the multiple labels and then cut. Currently doing the PDF to multiple PNG files with ghostscript.

pklaus, 2018-09-03:
Hello again, with the latest v0.9.x releases a new CLI came about: `brother_ql`. It allows printing multiple images at once (eg. using `brother_ql print -l 62 image_1.png image_2.png`). I'll see if I can further improve the printing speed for subsequent labels.

What concerns PDFs: The problem is rendering them to a pixel image. I two options: Using the [python bindings](https://pypi.org/project/ghostscript/) for [Ghostscript](https://www.ghostscript.com/) or the [python bindings](http://docs.wand-py.org/) for [ImageMagick](https://www.imagemagick.org). As I want to be cross-platform with brother_ql, I would consider supporting both. But I'd have to come up with a useful CLI, too. Any input would be welcome.

stuartlynne, 2018-09-03:
Unfortunately brother_ql print does not support sending the output to stdout.

I have other software that takes the resulting image data and sends it to a pool of QL printers. Currently three QL710W and two QL1060N printers. 

The application is printing bib (102x152) numbers and frame plate (62x100) numbers for bike races. For our largest cyclocross event we might print 500 labels on the QL1060N's and 2000 on the QL710W's over about four hours. Having multiple printers gets the labels out to the riders quickly and provides some redundancy.

pklaus, 2018-09-07:
> Unfortunately brother_ql print does not support sending the output to stdout.

With a rather recent change, the new brother_ql tool to supports specifying that an image file is provided via stdin using '-' as placeholder filename. That changeset is to be found in commit 0c59fa9.

You can upgrade to that development version (which is not yet part of a release) via:
pip install --upgrade https://github.com/pklaus/brother_ql/archive/0c59fa9.zip

pklaus, 2018-09-07:
And now back to the original topic of this issue: **Printing multiple "pages"**.

As mentioned in my [previous comment](https://github.com/pklaus/brother_ql/issues/24#issuecomment-418150339), the new CLI tool supports printing multiple image files in one go.

Please let me know if that speeds up the printing process as desired. I'll keep the issue open until I get some feedback from your side. If I find the time, I'll also compare the commands sent to the printer when using the brother_ql package (this software) to the ones sent by the Brother's P-touch Editor software (when printing multiple pages, of course).

stuartlynne, 2018-09-12:
The next issue is how to get multiple (e.g. two) labels printed with only a
but at the end?

resurrexi, 2019-12-21:
I'm wondering if there's an option to print multiple images using the `send` method. I'm not using the CLI since I want to use the core methods available in the package on a Django API app that prints barcodes.

Below is my utility script that my app uses. Note that you'll need `python-barcode==0.9.0` instead of the more recent version to get the code working.

```python
from io import BytesIO
from PIL import Image
from barcode.writer import ImageWriter
from barcode import generate
from brother_ql import BrotherQLRaster, create_label
from brother_ql.backends.helpers import send

__LABEL_OPTS = {
    '62': {
        'module_width': 0.13,
        'module_height': 8.0,
        'text_distance': 0.5,
        'font_size': 14,
        'dpi': 600,
        'quiet_zone': 1,
    },
    '29x90': {
        'module_width': 0.2,
        'module_height': 8.0,
        'text_distance': 0.5,
        'font_size': 14,
        'dpi': 600,
        'quiet_zone': 1
    },
}

# command: brother_ql info labels
__LABEL_PIXEL_SIZES = {
    '62': (696, 190),
    '29x90': (991, 306),
}

def print_barcode(data, printer, backend=None, model='QL-700', code='code128',
                  label='62'):
    """Generate barcode and print barcode with Brother QL printer.

    Parameters
    ----------
    data : str
        Text to encode to barcode
    printer : str
        Printer ID to send print instructions to. The value should be a string
        such as `tcp://192.168.0.100:9100` or `file:///dev/usb/lp0` or
        `usb://0x04f9:0x2042` depending on the backend used. The ID(s) can be
        found using `brother_ql discover`.
    backend : str, optional
        Printer backend.
    model : str
        The model of the printer
    code : str
        The barcode format to use. Refer to python-barcode documentation for
        available formats.
    label : str
        The label size. Refer to brother_ql documentation for available sizes.
    """
    # create image buffer object and save to object
    ib = BytesIO()
    generate(code, data, writer=ImageWriter(),
             writer_options=__LABEL_OPTS[label], output=ib)

    # center image to label dimensions
    old_img = Image.open(BytesIO(ib.getvalue()))
    old_size = old_img.size
    new_size = __LABEL_PIXEL_SIZES[label]
    # TODO: dynamically set `new_size` based on `label` argument
    new_img = Image.new(old_img.mode, new_size, 'white')
    new_img.paste(
        old_img,
        ((new_size[0] - old_size[0]) // 2, (new_size[1] - old_size[1]) // 2)
    )

    # send centered image to printer
    qlr = BrotherQLRaster(model)
    create_label(qlr, new_img, label, cut=True)
    send(
        qlr.data,
        printer_identifier=printer,
    )
```

michalmph, 2020-05-22:
I've managed to do that from python code. I hope this is what the library expects internally. I'm on Windows 10. 

```python
import os
import glob
import brother_ql
from brother_ql.raster import BrotherQLRaster
from brother_ql.backends.helpers import send

PRINTER_IDENTIFIER = 'tcp://192.168.88.202'
printer = BrotherQLRaster('QL-1050')

list_of_labels = []

# directory contains 3 png files
for filename in glob.iglob("./labels_to_print/" + '*', recursive=False):
    png_to_print = os.path.abspath(filename.replace('\\', '/'))
    list_of_labels.append(png_to_print)

# declare the print byte object
print_data = b''

def create_print_object(file_list):
    global print_data
    for i in file_list:
        print_data = print_data + (brother_ql.brother_ql_create.convert(printer, [i], '39x90', dither=True, cut=False))

create_print_object(list_of_labels)
        
# here the files are printed multiple times
send(print_data, PRINTER_IDENTIFIER, blocking=True)
```

**However, there are issues.** 
It seems I'm running into #63 - the printer prints the labels multiple times.

Additionally, this does not speed up the printing speed - it's exactly the same as printing from a loop.
The same goes for CLI using below command:

`brother_ql.exe --printer tcp://192.168.88.202 --model QL-1050 print --no-cut -l 39x90 '.\label_to_print_0 - Copy (2).png' '.\label_to_print_0 - Copy.png' .\label_to_print_0.png`

print speed is exactly the same with pauses between labels

CAVEATS:
- I'm on QL-1110NWB which is not listed as a supported printer. I may be able to test to QL-700 in the near future if there is need

johnlee999, 2020-06-19:
I use QL-820NWB.
As I really wanted to print multiple pages fast, I carefully re-read the brother's [raster command reference](https://download.brother.com/welcome/docp100278/cv_ql800_eng_raster_100.pdf ).

It seems that this library is not currently optimized for multi-page printing, probably for multi-model compatibility. 

I'm not yet familiar with git or Github, so I can't explain it well, but here are the major problems with printing multiple pages at the moment

**#1**

At the end of the instruction data on each page, you need a "print command".
There are two kinds of commands. 
1. Print command (0C) : Specifies at the end of a page That is not the last page.
2. Print command with feeding (1A) : Specifies at the end of the last page

Maybe because this library wasn't made to print multiple pages at first, this library always uses "1A" for each page.
I followed the reference, and the printing speed got so much faster.
To make it possible, I also needed to edit "--no-cut" option implementation. it should be controlled by two flags:
1. auto cut (Specify the page number in "cut each * labels")
2. cut at end

**#2**

The current method sends a "Status information request" for each page during printing. However, according to the reference, "Before sending print data to the printer, this command should be sent once. Since error
information is automatically sent by the printer during printing, do not send this command while printing."
I'm not sure if this made any difference though.

**#3**

As for "print information command", the ninth item {n9} should be "Starting page: 0 / Other pages: 1".
But the page count is not counted up in raster.py, so it is always 0, starting page.

****

There is also a problem with the --lq option. Because the argument is incorrectly determined, --lq / no --lq doesn't actually make any difference to the quality, it's always set to "high quality".
Printing in the original lq mode reduces the quality but dramatically increases the speed.

Now, I can print multi-page data very fast in my environment, as fast as p-touch editor does.
Once I understand a bit more about Github, I'll be able to make a pull request or something.

michalmph, 2020-06-23:
@johnlee999 That's great info, thanks a lot for this!
If time allows, I'll check this and report back.