pklaus/brother_ql #25: Response parse error

adaosantos, 2017-10-30:
OS: Ubuntu Xenial 16.04
Kernel: 4.10.0
Phyton: 2.7.12
PyUSB: 1.0.2
Printer: Brother QL-700
Command Line: sudo brother_ql_debug teste.bin usb://0x04f9:0x2042
[image]

Even with the above error it prints correctly.

pklaus, 2018-02-23:
Dear @adaosantos, thanks for reporting this issue. First of all, I would like to apologize I didn't respond earlier. I was assigning the issue a rather low priority as you stated that it prints after all. So I'm really happy that printing works, at least.
Your output shows that in one case only a small number of bytes is returned from the printer. In the next set of received bytes, a very large amount of data is received. In both cases, the info starts with `61 72 72 61 79 28 27 42 27`. Usually, the status received from the printers is exactly 32 bytes long with the first three bytes being `80 20 42`. So I really have no clue what the printer returned here.
The only idea would to try again. Maybe with using the latest version of *brother_ql* (upgrade using `pip install --upgrade brother_ql`)? Or with a different USB cable?

jpontius, 2018-03-15:
Hello,  I am also seeing this `Couldn't interpret response` problem.  I'm trying to use DK-2214 labels with QR codes.  I'll attach the files I'm trying to use. 

**Linux label-pi 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l GNU/Linux**

```
brother_ql_debug 0000.bin /dev/usb/lp0 
INFO: CMD mode setting FOUND. Instruction: 1B 69 61 01 
INFO: CMD preamble FOUND. Instruction: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [...] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
INFO: CMD init FOUND. Instruction: 1B 40 
INFO: CMD mode setting FOUND. Instruction: 1B 69 61 01 
INFO: CMD status request FOUND. Instruction: 1B 69 53 
INFO: Response from the device: 80 20 42 34 38 30 00 00 00 00 0C 0A 00 00 1A 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00
ERROR: Couln't interpret response: 80 20 42 34 38 30 00 00 00 00 0C 0A 00 00 1A 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00
INFO: CMD media/quality FOUND. Instruction: 1B 69 7A CE 0B 0C 0C 5E 00 00 00 00 00 
INFO: CMD various FOUND. Instruction: 1B 69 4D 40 
INFO: CMD cut-every FOUND. Instruction: 1B 69 41 01 
INFO: CMD expanded FOUND. Instruction: 1B 69 4B 08 
INFO: CMD margins FOUND. Instruction: 1B 69 64 23 00 
INFO: CMD raster FOUND. Instruction: 67 00 5A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [...] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
INFO: CMD print FOUND. Instruction: 1A 
INFO: Response from the device: 80 20 42 34 38 30 00 00 00 00 0C 0A 00 00 1A 00 00 00 06 01 00 00 00 00 00 01 00 00 00 00 00 00
ERROR: Couln't interpret response: 80 20 42 34 38 30 00 00 00 00 0C 0A 00 00 1A 00 00 00 06 01 00 00 00 00 00 01 00 00 00 00 00 00
TIME 1.38
ERROR: Couln't interpret response: 80 20 42 34 38 30 00 00 00 01 0C 0A 00 00 1A 00 00 00 02 01 00 00 00 00 00 01 00 00 00 00 00 00
```
[image]
[image]
[bin.zip](https://github.com/pklaus/brother_ql/files/1816280/bin.zip)

AngelSotomayor, 2018-07-30:
I see the same error code:

WARNING: Received response before sending instructions: 61 72 72 61 79 28 27 42 27 29
TIME 0.00
ERROR: Couln't interpret response: 61 72 72 61 79 28 27 42 27 29
TIME 0.00
ERROR: Couln't interpret response: 61 72 72 61 79 28 27 42 27 29

Followed by this:

INFO: CMD preamble FOUND. Instruction: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [...] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
INFO: Response from the device: 61 72 72 61 79 28 27 42 27 29
ERROR: Couln't interpret response: 61 72 72 61 79 28 27 42 27 29
INFO: CMD init FOUND. Instruction: 1B 40
INFO: Response from the device: 61 72 72 61 79 28 27 42 27 29
ERROR: Couln't interpret response: 61 72 72 61 79 28 27 42 27 29
INFO: CMD status request FOUND. Instruction: 1B 69 53
INFO: Response from the device: 61 72 72 61 79 28 27 42 27 2C 20 5B 31 32 38 2C 20 33 32 2C 20 36 36 2C 20 35 32 2C 20 35 33 2C 20 34 38 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 36 32 2C 20 31 30 2C 20 30 2C 20 30 2C 20 32 31 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 5D 29
ERROR: Couln't interpret response: 61 72 72 61 79 28 27 42 27 2C 20 5B 31 32 38 2C 20 33 32 2C 20 36 36 2C 20 35 32 2C 20 35 33 2C 20 34 38 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 36 32 2C 20 31 30 2C 20 30 2C 20 30 2C 20 32 31 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 5D 29
INFO: CMD media/quality FOUND. Instruction: 1B 69 7A CE 0D 0A 3E 00 91 00 00 00 00
INFO: Response from the device: 61 72 72 61 79 28 27 42 27 29
ERROR: Couln't interpret response: 61 72 72 61 79 28 27 42 27 29
INFO: CMD preamble FOUND. Instruction: 00
INFO: Response from the device: 61 72 72 61 79 28 27 42 27 29
ERROR: Couln't interpret response: 61 72 72 61 79 28 27 42 27 29
INFO: CMD margins FOUND. Instruction: 1B 69 64 23 00
INFO: Response from the device: 61 72 72 61 79 28 27 42 27 29
ERROR: Couln't interpret response: 61 72 72 61 79 28 27 42 27 29
INFO: CMD raster FOUND. Instruction: 67 00 5A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0F FF FF F1 FF FF FF FF FF FE 00 00 00 00 00 07 FF FF 03 FF FF FF FF C0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [...] 00 FF FF E0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
INFO: Response from the device: 61 72 72 61 79 28 27 42 27 2C 20 5B 31 32 38 2C 20 33 32 2C 20 36 36 2C 20 35 32 2C 20 35 33 2C 20 34 38 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 36 32 2C 20 31 30 2C 20 30 2C 20 30 2C 20 32 31 2C 20 30 2C 20 30 2C 20 30 2C 20 36 2C 20 31 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 5D 29
ERROR: Couln't interpret response: 61 72 72 61 79 28 27 42 27 2C 20 5B 31 32 38 2C 20 33 32 2C 20 36 36 2C 20 35 32 2C 20 35 33 2C 20 34 38 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 36 32 2C 20 31 30 2C 20 30 2C 20 30 2C 20 32 31 2C 20 30 2C 20 30 2C 20 30 2C 20 36 2C 20 31 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 5D 29
INFO: CMD print FOUND. Instruction: 1A
INFO: Response from the device: 61 72 72 61 79 28 27 42 27 29
ERROR: Couln't interpret response: 61 72 72 61 79 28 27 42 27 29
TIME 0.57
ERROR: Couln't interpret response: 61 72 72 61 79 28 27 42 27 29

And like that for 3 seconds and nothing is printed, any clue?

mapadj, 2019-10-19:
Hello. I also receive this response on a QL-800 with latest brother_ql library.

`brother_ql --backend pyusb --model QL-800 --printer usb://0x04f9:0x209b/000E9Z943257 print -l 62x29 label.png
No handlers could be found for logger "brother_ql.devicedependent"
INFO:brother_ql.backends.helpers:Sending instructions to the printer. Total: 25447 bytes.
Traceback (most recent call last):
  File "/usr/local/bin/brother_ql", line 10, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 696, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 1060, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/brother_ql/cli.py", line 147, in print_cmd
    send(instructions=instructions, printer_identifier=printer, backend_identifier=backend, blocking=True)
  File "/usr/local/lib/python2.7/site-packages/brother_ql/backends/helpers.py", line 76, in send
    result = interpret_response(data)
  File "/usr/local/lib/python2.7/site-packages/brother_ql/reader.py", line 159, in interpret_response
    raise NameError("Printer response doesn't start with the usual header (80:20:42)", hex_format(data))
NameError: ("Printer response doesn't start with the usual header (80:20:42)", '61 72 72 61 79 28 27 42 27 2C 20 5B 31 32 38 2C 20 33 32 2C 20 36 36 2C 20 35 32 2C 20 35 36 2C 20 34 38 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 36 32 2C 20 31 30 2C 20 30 2C 20 30 2C 20 33 35 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 31 32 39 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 5D 29')
`

I am trying all day, but i cannot figure out, whats wrong. I can print with P-Touch, but i can't get the printer to print via the standard print dialog not via the python library.

pklaus, 2020-02-04:
This should be fixed now. Wow, it took me from Oct 30, 2017 to now to fix this.
The problem: It was a Python 2.x bug and I don't use Python 2 personally.

Thanks for reporting nevertheless. Please let me know if the problem persists with the current version from Github. I'll publish a new release in the next days.

commit 56cf4394 (2020-02-04): [fix] "response doesn't start with the usual header"

fixes #25, fixes #69, fixes #75, fixes #81

also seen in issues: #56, #71
closed 2020-02-04 by pklaus