pklaus/brother_ql #27: Extra Label Size

i6media, 2017-11-22:
Hi,

Thanks for your awesome package, works like a charm!

There is however one thing, we've purchased a Brother QL-1060N, with address labels. The format looks like it's 102x152 mm, but it is 102x153 mm.

That means it flashes red when printing in the 152x102 format.

We've modified devicedependent.py ourselves, and added the extra label size ourselves, however if other people experience the same problem, permanent implementation in the repository would be great:

`"102x153":{'tape_size': (102, 153), 'dots_total': (1200, 1804), 'dots_printable': (1164, 1660), 'right_margin_dots': 12, 'feed_margin':  0},`

Keep up the good work!

pklaus, 2017-11-23:
Great, thanks for this report. I'm happy someone can confirm the QL-1060N to work.
So printing works nicely with the added line? Did you try any other label sizes, too?

i6media, 2017-11-23:
I didn't try other formats, but I did investigate some more:

Even on the cardboard package (Brother DK-11241) it says 102x152mm Address Label, however on the printer verbose log it says 'currently installed: 102x153mm'.

With the added line on devicedependent.py it works like a charm.

pklaus, 2017-11-23:
Good, thanks for letting me know.

If Brother is advertising it that way, I think I should change the label specification line to state `'tape_size': (102, 153),` but leave the label identifier (for the command line) as `102x152`.

pklaus, 2018-01-06:
Commit e5d4fde should fix this issue. I kept the previous label-size identifier `"102x152"`. Behind the scenes, however, the package is now using the 102x153 info. Thanks for reporting this!

commit e5d4fde7 (2018-01-06): Correct Brother DK-11241 label size "102x152", fixes #27
closed 2018-01-06 by pklaus