pklaus/brother_ql #28: ERROR: Couln't interpret response from QL-1050

youpla, 2018-01-05:
Hello,

I'm trying to print to my QL-1050 on a 102x152 sticker.
When the job is sent, the printer LED turns blinking red.
Looking at the DEBUG Doc I was able to get the following information. 
It looks like some response cannot be interpreted. 
If you need further information, just tell me which command to run.
Thanks

`brother_ql_debug 1164_1660_monochrome.bin /dev/usb/lp0
INFO: CMD mode setting FOUND. Instruction: 1B 69 61 01
INFO: CMD preamble FOUND. Instruction: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0                                                                0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00                                                                 00 00 00 00 [...] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
INFO: CMD init FOUND. Instruction: 1B 40
INFO: CMD mode setting FOUND. Instruction: 1B 69 61 01
INFO: CMD status request FOUND. Instruction: 1B 69 53
INFO: Response from the device: 80 20 42 30 50 30 00 00 00 00 66 0B 00 00 2E 00 00 99 00 00 00 00 00 00 00 00 00 00 00                                                                 00 00 00
ERROR: Couln't interpret response: 80 20 42 30 50 30 00 00 00 00 66 0B 00 00 2E 00 00 99 00 00 00 00 00 00 00 00 00 00                                                                 00 00 00 00
INFO: CMD media/quality FOUND. Instruction: 1B 69 7A CE 0B 66 98 7C 06 00 00 00 00
INFO: CMD various FOUND. Instruction: 1B 69 4D 40
INFO: CMD cut-every FOUND. Instruction: 1B 69 41 01
INFO: CMD expanded FOUND. Instruction: 1B 69 4B 08
INFO: CMD margins FOUND. Instruction: 1B 69 64 00 00
INFO: CMD raster FOUND. Instruction: 67 00 A2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [...] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
INFO: Response from the device: 80 20 42 30 50 30 00 00 00 00 66 0B 00 00 2E 00 00 99 06 01 00 00 00 00 00 00 00 00 00 00 00 00
ERROR: Couln't interpret response: 80 20 42 30 50 30 00 00 00 00 66 0B 00 00 2E 00 00 99 06 01 00 00 00 00 00 00 00 00 00 00 00 00
INFO: CMD print FOUND. Instruction: 1A
INFO: Response from the device: 80 20 42 30 50 30 00 00 00 01 66 0B 00 00 2E 00 00 99 02 01 00 00 00 00 00 00 00 00 00 00 00 00
ERROR: Couln't interpret response: 80 20 42 30 50 30 00 00 00 01 66 0B 00 00 2E 00 00 99 02 01 00 00 00 00 00 00 00 00 00 00 00 00`

pklaus, 2018-01-06:
Thanks for reporting this issue. The problem is that the label is called 102x152 by Brother but behind the scenes, it's actually 102x153. This was reported before in issue #27. I released a new version which should fix this issue: [brother_ql v0.8.3](https://github.com/pklaus/brother_ql/releases/tag/). 
Please upgrade (using `pip install --upgrade brother_ql`) and report if this works now.
Note, that I kept the identifier for the `--label-size` as it was before: `102x152`.

Thanks for your bug report!

youpla, 2018-01-07:
Hi Klaus,
Update has been installed and works great.
Thanks for making this software, my rpi is now printing to my QL-1050 like a charm.

Mit freundlichen Grüssen

closed 2018-01-07 by youpla