pklaus/brother_ql #31: Works from cli, not from script

luisrsiu, 2018-02-22:
Need help please, dont know what am i doing wrong. The following line:

brother_ql_create --model QL-710W /home/administrador/etiquetas/etiqueta.png > /home/administrador/etiquetas/etiqueta_monochrome.bin

works when running the script from the command line, but outputs a file of length 0 when called by a cron script. Thank you.

pklaus, 2018-02-22:
As a first idea: Check, where brother_ql_create is installed with `which brother_ql_create` and provide that info here. Try to use that absolute path in your your crontab.

luisrsiu, 2018-02-23:
Did what you suggested, by using the absolute path it works like a charm. Thank you very much for your help, really appreciate it; the project is already up and running!

pklaus, 2018-02-23:
I'm happy I was able to help.
If you can make good use of the project, I would be glad to hear what you use it for and what you like about it. You could send me an email (address to be found on my github account page: <https://github.com/pklaus>). Thanks!

closed 2018-02-27 by pklaus