pklaus/brother_ql #32: Windows support

benmoriz, 2018-05-17:
Hello,

I am able to create an *.bin file from image but can't print it.

Dose **brother_ql_print *.bin usb://0x04f9:0x2015**  support for python on windows
. 
I am trying a lot but I keep on getting,

Traceback (most recent call last):
  File "c:\users\nischal\appdata\local\programs\python\python37\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\nischal\appdata\local\programs\python\python37\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\Nischal\AppData\Local\Programs\Python\Python37\Scripts\brother_ql_print.exe\__main__.py", line 9, in <module>
  File "c:\users\nischal\appdata\local\programs\python\python37\lib\site-packages\brother_ql\brother_ql_print.py", line 82, in main
    printer = BrotherQLBackend(string_descr)
  File "c:\users\nischal\appdata\local\programs\python\python37\lib\site-packages\brother_ql\backends\pyusb.py", line 76, in __init__
    for result in list_available_devices():
  File "c:\users\nischal\appdata\local\programs\python\python37\lib\site-packages\brother_ql\backends\pyusb.py", line 43, in list_available_devices
    printers = usb.core.find(find_all=1, custom_match=find_class(7), idVendor=0x04f9)
  File "c:\users\nischal\appdata\local\programs\python\python37\lib\site-packages\usb\core.py", line 1263, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available 

Any response will be highly appreciated.

mampez, 2018-05-21:
I got the same error msg when I tried to print at Windows... "No backend availabe".

mampez, 2018-05-21:
You need to install this before printing: https://sourceforge.net/projects/libusb-win32/files/libusb-win32-releases/1.2.6.0/

benmoriz, 2018-05-22:
It is working perfectly on command prompt and I am able to print now. 
Thank you. 
But, do you have any idea how can I create an instruction file and print on Python IDLE because it doesn't recognize model, I get syntax error on 'QL'.

pklaus, 2018-05-22:
Great to hear that your "printing on Windows" issue is solved. I should probably put this info in the README.

What concerns your other issue (with Python IDLE): Would you mind opening a separate issue? Probably including the full error (traceback) and how you call it.

Thanks for showing your interest in the project and for opening the issue.

closed 2018-05-22 by pklaus