pklaus/brother_ql #33: Creating instruction file on script and printing on QL-800

benmoriz, 2018-05-23:
The following code works perfectly (creates abc.bin file and prints from the printer) on command prompt:
_brother_ql_create --model QL-800 --red abc.png --label-size 62> abc.bin
brother_ql_print abc.bin usb://0x04F9:0x209B_

On python script:
_from brother_ql import brother_ql_create
from brother_ql import brother_ql_print
from PIL.ImageTk import PhotoImage

brother_ql_create --model QL-800 --red abc.png --label-size 62> abc.bin_

Error: 'QL' is invalid syntax__ 

Why is it not possible to call the following line on python script?
Is there any other method to create instruction file and print from script?

pklaus, 2018-05-23:
Dear benmoriz,

you cannot use the Shell line inside a Python script. But you can create a script that imports the right modules and classes, create the label data and print it:

```python
# creating the label
from brother_ql import create_label, BrotherQLRaster
qlr = BrotherQLRaster("QL-800")
create_label(qlr, your_image, "62", threshold=70, cut=True, red=True)

# sending data to printer
from brother_ql.backends import backend_factory, guess_backend, available_backends
selected_backend = guess_backend("usb://0x04F9:0x209B")
be = backend_factory(selected_backend)
BrotherQLBackend       = be['backend_class']
printer = BrotherQLBackend(string_descr)
# Sending instructions to the printer.
printer.write(qlr.data)
```
The create_label() function is the central piece here. See [this](https://github.com/pklaus/brother_ql/blob/master/brother_ql/brother_ql_create.py#L65) for its implementation.

Afterwards, the data to be sent to the printer is to be found in the property `qlr.data`. See [brother_ql_print.py](https://github.com/pklaus/brother_ql/blob/master/brother_ql/brother_ql_print.py) for a complete 'printing' implementation.

benmoriz, 2018-05-24:
Hi Klaus,
Thank you for your instruction, it works.
And a great library, 
it's much more interesting to try out different things from it. 

Never tried using printer,  
but I can do the printing job very well now.   
I'll definitely share my work with you after I complete it's development. 

Thanks again.

closed 2018-05-24 by benmoriz

achtha, 2019-03-11:
Hi,
I am trying above code to print from Ql-820NWB printer through Windows 10 OS. I get error as below:
deprecation warning: brother_ql.devicedependent is deprecated and will be removed in a future release
Need to resize the image...
Traceback (most recent call last):
  File "D:/Programming/Python/Printer Python/QLPrinterTest.py", line 10, in <module>
    printer = BrotherQLBackend('usb://0x04f9:0x2015/C5Z315686') # string_descr = ???
  File "C:\Users\Local\Programs\Python\Python37\lib\site-packages\brother_ql\backends\pyusb.py", line 79, in __init__
    for result in list_available_devices():
  File "C:\Users\Local\Programs\Python\Python37\lib\site-packages\brother_ql\backends\pyusb.py", line 46, in list_available_devices
    printers = usb.core.find(find_all=1, custom_match=find_class(7), idVendor=0x04f9)
  File "C:\Users\Local\Programs\Python\Python37\lib\site-packages\usb\core.py", line 1263, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available

My actual code is:

from brother_ql import create_label, BrotherQLRaster
qlr = BrotherQLRaster("QL-820NWB")
create_label(qlr, r'D:\Programming\Python\Printer Python\Image\newImg.png', "62",
             threshold = 70, cut=True, red=True)
from brother_ql.backends import backend_factory, guess_backend, available_backends
selected_backend = guess_backend("usb://0x04F9:0x209B")
be = backend_factory(selected_backend)
BrotherQLBackend = be['backend_class'] # backend_class =  ?? 
printer = BrotherQLBackend('usb://0x04f9:0x2015/C5Z315686') # string_descr = ???
printer.write(qlr.data)

What i am missing kind of lost as i am trying to connect dots to make a successful print.