pklaus/brother_ql #34: Getting errors when list-backends

kevinsimper, 2018-06-19:
I get this error, I have installed `brew install libusb` and `pip install pyusb`

```
$ brother_ql_print --backend pyusb --list-printers
Traceback (most recent call last):
  File "/usr/local/bin/brother_ql_print", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/site-packages/brother_ql/brother_ql_print.py", line 37, in main
    with open(args.instruction_file, 'rb') as f:
TypeError: expected str, bytes or os.PathLike object, not NoneType
```

kevinsimper, 2018-06-19:
Got it only working with Bluetooth before, but I got a lot of problems with unreliable delivery, a lot of "Receiving data" getting stuck.

Other problem was I did not know anything about usb and the `--list-printers` did not work and did not want to use Wifi.

Googled and found out you can do this on mac to list usb devices but it does not show the address the way pyusb wants it:
```
$ ioreg -p IOUSB -w0 -l
```

Figured out to see if pyusb could just list it and that worked
```
$ python3
Python 3.6.5 (default, Jun 17 2018, 12:26:58)
[GCC 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.39.2)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb.core
>>> dev = usb.core.find()
dev
>>> dev
<DEVICE ID 04f9:209d on Bus 020 Address 031>
```

and then I could do

```
$ brother_ql_debug label.bin usb://04f9:209d
INFO: CMD mode setting FOUND. Instruction: 1B 69 61 01
INFO: CMD preamble FOUND. Instruction: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [...] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
INFO: CMD init FOUND. Instruction: 1B 40
INFO: CMD mode setting FOUND. Instruction: 1B 69 61 01
INFO: CMD status request FOUND. Instruction: 1B 69 53
INFO: Response from the device: 80 20 42 34 41 30 04 00 00 00 3E 0A 00 00 15 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00
INFO: Interpretation of the response: 'Reply to status request' (phase: Waiting to receive), 'Continuous length tape' 62x0 mm^2, errors: []
INFO: CMD media/quality FOUND. Instruction: 1B 69 7A CE 0A 3E 00 82 03 00 00 00 00
INFO: CMD various FOUND. Instruction: 1B 69 4D 40
INFO: CMD cut-every FOUND. Instruction: 1B 69 41 01
INFO: CMD expanded FOUND. Instruction: 1B 69 4B 08
INFO: CMD margins FOUND. Instruction: 1B 69 64 23 00
INFO: CMD raster FOUND. Instruction: 67 00 5A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [...] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
INFO: Response from the device: 80 20 42 34 41 30 04 00 00 00 3E 0A 00 00 15 00 00 00 06 01 00 00 00 00 00 01 00 00 00 00 00 00
INFO: Interpretation of the response: 'Phase change' (phase: Printing state), 'Continuous length tape' 62x0 mm^2, errors: []
INFO: CMD print FOUND. Instruction: 1A
TIME 3.45
INFO: Interpretation of the response: 'Printing completed' (phase: Printing state), 'Continuous length tape' 62x0 mm^2, errors: []
TIME 3.46
INFO: Interpretation of the response: 'Phase change' (phase: Waiting to receive), 'Continuous length tape' 62x0 mm^2, errors: []
```

and it works so much faster than Bluetooth.

pklaus, 2018-08-09:
Hello @kevinsimper, thanks for reporting.

Your initial issue with running `brother_ql_print --backend pyusb --list-printers` is that the brother\_ql\_print needs an argument. Even if you're just 'discovering' devices and not printing. Yes this is stupid, sorry. To fix this and many other issues, I just released a version which has a new CLI tool. Now, you can run: `brother_ql --backend pyusb discover`. Please report if there are issues with that command.

Besides that, what exactly do you mean by *no way to get the current version installed*? Is the command `pip install --upgrade brother_ql` failing?

> Got it only working with Bluetooth before, but I got a lot of problems with unreliable delivery, a lot of "Receiving data" getting stuck.

Interesting. How did you use the Bluetooth interface? With Brother's drivers? On your phone? Or with this software package brother_ql (if that's possible at all)?

I'm glad you could make it print in the end.

Thanks for reporting.

kevinsimper, 2018-08-09:
> Besides that, what exactly do you mean by no way to get the current version installed?

There is missing a command like `brother_ql --version` 😄 

> Interesting. How did you use the Bluetooth interface? 

`cat label.bin > /dev/tty.QL-820NWB9635-SerialPort`

> With Brother's drivers?

I actually did it without drivers installed on Mac

> Or with this software package brother_ql (if that's possible at all)?

After I found the pyusb address I use the `brother_ql_print` and it works 100%. It worked without the drivers, but with the drivers installed the print is much quicker.

pklaus, 2018-08-10:
> `cat label.bin > /dev/tty.QL-820NWB9635-SerialPort`

Interesting. Didn't work with the bluetooth connection before and I wouldn't have thought it simply appears as a *serial port*. I guess, the WiFi / Ethernet / USB connectivity is still the better choice, though.

> There is missing a command like brother_ql --version 😄

Alright. I added `brother_ql --version` to the current development version in the master branch.

> After I found the pyusb address I use the brother_ql_print and it works 100%. It worked without the drivers, but with the drivers installed the print is much quicker.

What *drivers* are you refering to here?

pklaus, 2018-08-10:
Now there's also a new command showing even more information:

    brother_ql info env

It was introduced in 78b5667 and its output is going to be helpful for any future bug reports.

kevinsimper, 2018-08-10:
> What drivers are you refering to here?

You can find the drivers here! By no means am I a driver expert so I did nothing other than install it. 

http://support.brother.com/g/b/downloadlist.aspx?c=us&lang=en&prod=lpql820nwbeus&os=10046

> Now there's also a new command showing even more information:

Thank you, awesome!

kevinsimper, 2018-08-10:
@pklaus Okay interesting, thanks for clearing that up! 😄