pklaus/brother_ql #35: Python 2.7 compatibility

spidfire, 2018-07-10:
> NOTE: Do not merge without a 3.0 fix  (it's a 2.7 only fix)

https://github.com/spidfire/brother_ql/blob/master/brother_ql/reader.py

during the use of brother_ql_print it got an error of unable to convert str to int

I've created a python 2.7 fix (it doesn't work with py3+) 

it fixes the problem of that a bytearray element like `data[1]` returns a string in 2.7 and a int in 3+

pklaus, 2018-07-27:
Thanks for sharing those changes. I'll do some testing on Py3 in order to ensure compatibility with both. Probably the minimum 3.x version I'll support is 3.3 or 3.4. Might take me another week to find the time to complete testing.

spidfire, 2018-07-27:
Nice thx :+1:

closed 2018-08-08 by pklaus

pklaus, 2018-08-08:
The required change in fd22068 was even simpler.
That's the advantage of using the `from builtins import bytes` import statement provided by the [future](https://pypi.org/project/future/) Py2/Py3 compatibility layer package.

Thanks again.

commit fd220689 (2018-08-08): Py2 compat of brother_ql.reader - fixes #35