pklaus/brother_ql #36: Installation Error

zachtheriah, 2018-07-25:
When installing the software, I have found that there is a problem right after installation: the brother_ql_* programs do not run from the command line. Is there something else I should be doing?

[image]

zachtheriah, 2018-07-25:
(Is it possible that continuous length tape is the problem?)

zachtheriah, 2018-07-25:
Found the problem; had to change the create size option from 62x29 to 62. Thanks for building this great software! :)

closed 2018-07-25 by zachtheriah

pklaus, 2018-07-27:
Hi @zachtheriah, thanks for sharing your issues. As a matter of fact, those are the most frequently made mistakes when using this software:

1. Using a pip flavour that installs the executables into the `~/.local/bin` folder, which is not in the PATH environment variable by default.
2. Creating labels without specifying the correct `--model`.
3. Creating labels without specifying the correct `--label-size`.

As this occured multiple times, I guess I should point out those things more clearly in the instructions (README) or create a Youtube video about 'getting started'.

Your input is valuable for me to see where I can improve the software. Thanks again for letting me know.