pklaus/brother_ql #40: Error: Printer can be specified either String or os open call

hydroid7, 2018-08-25:
I set up my env and in terminal I did:
```
(env) $ export BROTHER_QL_PRINTER=file:///dev/usb/lp0
(env) $ export BROTHER_QL_MODEL=QL-700
(env) $ brother_ql print -l 62 my_image.png
``` 
But every time I get ` NotImplementedError('Currently the printer can be specified either via an appropriate string or via an os.open() handle.')`

 https://github.com/pklaus/brother_ql/blob/fe55344d32e41ba2428b9b4f1198d018d8873132/brother_ql/backends/linux_kernel.py#L46

I can verify that printing works because I tested it with the legacy CLI tools.

christianflintrup, 2018-08-27:
Experiencing the same issue. Also tried passing them on the command-line rather than environment variables:

`brother_ql --printer tcp://192.168.1.142 --model QL-1060N print -l 102x152 test.png`

Results in:

`NotImplementedError: Currently the printer can be specified either via an appropriate string or via an os.open() handle.`

Using default Python 2.7.13 bundled with Mac OS X.

uched41, 2018-08-31:
I had the same issue a few days ago, it seems for some reason the script is not recognizing the exported variables, try passing them as command line variables:

sudo brother_ql -p usb://04f9:0242 -m QL-1060N print -l 102x152 test.png

Note: the size is very important if you do not know the size, you can try ` -l 62 ` it seems to work with different sizes. Also make sure that you have pyusb installed.

pip install pyusb

You may need to run the command as sudo.

pklaus, 2018-09-05:
As 7ab1e67 is now part of the latest release v0.9.2, which can be installed using `pip install --upgrade brother_ql`, I'll close this issue. Thanks again for reporting.

closed 2018-09-05 by pklaus