pklaus/brother_ql #44: ValueError: invalid literal for int() with base 16: 'Brother'

j99ht, 2018-09-28:
I was trying to print with my Brother QL810W in Ubuntu.
`brother_ql print -l 62 label.png`

The driver works fine but when I try to use brother_ql I get this error

```
Traceback (most recent call last):
  File "/home/myhome/.local/bin/brother_ql", line 11, in <module>
    sys.exit(cli())
  File "/home/myhome/.local/lib/python2.7/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/myhome/.local/lib/python2.7/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/myhome/.local/lib/python2.7/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/myhome/.local/lib/python2.7/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/myhome/.local/lib/python2.7/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/myhome/.local/lib/python2.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/myhome/.local/lib/python2.7/site-packages/brother_ql/cli.py", line 147, in print_cmd
    send(instructions=instructions, printer_identifier=printer, backend_identifier=backend, blocking=True)
  File "/home/communithings/.local/lib/python2.7/site-packages/brother_ql/backends/helpers.py", line 57, in send
    printer = BrotherQLBackend(printer_identifier)
  File "/home/myhome/.local/lib/python2.7/site-packages/brother_ql/backends/pyusb.py", line 78, in __init__
    vendor, product = int(vendor, 16), int(product, 16)
ValueError: invalid literal for int() with base 16: 'Brother'
```

I have these exports
export BROTHER_QL_PRINTER=usb://Brother/QL-810W?serial=000L7Z460260
export BROTHER_QL_MODEL=QL-810W

Am I doing anything wrong?

pklaus, 2018-09-28:
Thanks for the report. I'll investigate it. Could you please post the output of the command `brother_ql info env` in addition?

pklaus, 2018-09-28:
Actually this line is the culprit:

    export BROTHER_QL_PRINTER=usb://Brother/QL-810W?serial=000L7Z4xxxxx

You shouln't use `usb://Brother/QL-810W`. The format to use here is of the form `usb://0x1234:0x5678` where 0x1234 stands for the hexadecimal vendor id and 0x5678 stands for the hexadecimal product id. I'm considering to support the format (that Mac OS is using / defining?? and that you were trying to use) in the future.

To find the right string to use, you may also run:

```
export BROTHER_QL_BACKEND=pyusb
brother_ql discover
```