pklaus/brother_ql #45: Call brother_ql from PHP

j99ht, 2018-10-01:
Is there is a way to call brother_ql from a php script?

I'm trying it but I get this error

`Traceback (most recent call last): File "/usr/bin/brother_ql", line 7, in from brother_ql.cli import cli ImportError: No module named brother_ql.cli
`

pklaus, 2018-10-01:
You can without a doubt use the brother_ql tool from other software and many do this succesfully. Could you please state how you were installing brother_ql and paste the output of running the following command in the terminal:

    brother_ql info env

pklaus, 2018-10-01:
Thanks for posting the status output. It shows that you installed the package in your user directory:

    package location: /home/myhome/.local/lib/python2.7/site-packages

The reason being that `pip install somepackage` defaults to `pip install --user somepackage` on Ubuntu.

Please consider installing the package system wide with `sudo pip install brother_ql` or run your PHP server under the same user as the one you installed the package with `pip` on the command line.

closed 2018-10-01 by pklaus

pklaus, 2018-10-02:
So just to check: Did this solve the issue for you?

j99ht, 2018-10-02:
It worked. Thanks a lot

pklaus, 2018-10-02:
OK, great. Let me know if there is something you think could be improved (ideally in a new issue or via email).