pklaus/brother_ql #46: Printing aborts on QL-700

rec0de, 2018-10-17:
Hi there!
I just found this awesome tool after losing my mind over brother drivers for a few hours and ran into some trouble with my QL-700 printer.
Printing a label fails somewhere halfway with hardware buttons becoming unresponsive until I restart the printer. The issue seems to be very similar to #11 from what I can tell.

Here's a debug log of the print job:
```DEBUG:brother_ql.raster:raster_image_size: 720x696
INFO:brother_ql.backends.helpers:Sending instructions to the printer. Total: 64964 bytes.
DEBUG:brother_ql.backends.generic:Writing 64964 bytes.
DEBUG:brother_ql.backends.generic:Read 32 bytes.
DEBUG:brother_ql.reader:Byte  0         Print head mark: 80
DEBUG:brother_ql.reader:Byte  1                    Size: 20
DEBUG:brother_ql.reader:Byte  2          Fixed (B=0x42): 42
DEBUG:brother_ql.reader:Byte  3        Device dependent: 34
DEBUG:brother_ql.reader:Byte  4        Device dependent: 35
DEBUG:brother_ql.reader:Byte  5          Fixed (0=0x30): 30
DEBUG:brother_ql.reader:Byte  6  Fixed (0x00 or 0=0x30): 00
DEBUG:brother_ql.reader:Byte  7            Fixed (0x00): 00
DEBUG:brother_ql.reader:Byte  8     Error information 1: 00
DEBUG:brother_ql.reader:Byte  9     Error information 2: 00
DEBUG:brother_ql.reader:Byte 10             Media width: 3E
DEBUG:brother_ql.reader:Byte 11              Media type: 0A
DEBUG:brother_ql.reader:Byte 12            Fixed (0x00): 00
DEBUG:brother_ql.reader:Byte 13            Fixed (0x00): 00
DEBUG:brother_ql.reader:Byte 14                Reserved: 15
DEBUG:brother_ql.reader:Byte 15                    Mode: 00
DEBUG:brother_ql.reader:Byte 16            Fixed (0x00): 00
DEBUG:brother_ql.reader:Byte 17            Media length: 00
DEBUG:brother_ql.reader:Byte 18             Status type: 00
DEBUG:brother_ql.reader:Byte 19              Phase type: 00
DEBUG:brother_ql.reader:Byte 20     Phase number (high): 00
DEBUG:brother_ql.reader:Byte 21      Phase number (low): 00
DEBUG:brother_ql.reader:Byte 22     Notification number: 00
DEBUG:brother_ql.reader:Byte 23                Reserved: 00
DEBUG:brother_ql.reader:Byte 24                Reserved: 00
DEBUG:brother_ql.reader:Media type: Continuous length tape
DEBUG:brother_ql.reader:Status type: Reply to status request
DEBUG:brother_ql.reader:Phase type: Waiting to receive
DEBUG:brother_ql.backends.helpers:TIME 5.026 - result: {'status_type': 'Reply to status request', 'phase_type': 'Waiting to receive', 'media_type': 'Continuous length tape', 'media_width': 62, 'media_length': 0, 'errors': []}
DEBUG:brother_ql.backends.generic:Read 32 bytes.
DEBUG:brother_ql.reader:Byte  0         Print head mark: 80
DEBUG:brother_ql.reader:Byte  1                    Size: 20
DEBUG:brother_ql.reader:Byte  2          Fixed (B=0x42): 42
DEBUG:brother_ql.reader:Byte  3        Device dependent: 34
DEBUG:brother_ql.reader:Byte  4        Device dependent: 35
DEBUG:brother_ql.reader:Byte  5          Fixed (0=0x30): 30
DEBUG:brother_ql.reader:Byte  6  Fixed (0x00 or 0=0x30): 00
DEBUG:brother_ql.reader:Byte  7            Fixed (0x00): 00
DEBUG:brother_ql.reader:Byte  8     Error information 1: 00
DEBUG:brother_ql.reader:Byte  9     Error information 2: 00
DEBUG:brother_ql.reader:Byte 10             Media width: 3E
DEBUG:brother_ql.reader:Byte 11              Media type: 0A
DEBUG:brother_ql.reader:Byte 12            Fixed (0x00): 00
DEBUG:brother_ql.reader:Byte 13            Fixed (0x00): 00
DEBUG:brother_ql.reader:Byte 14                Reserved: 15
DEBUG:brother_ql.reader:Byte 15                    Mode: 00
DEBUG:brother_ql.reader:Byte 16            Fixed (0x00): 00
DEBUG:brother_ql.reader:Byte 17            Media length: 00
DEBUG:brother_ql.reader:Byte 18             Status type: 06
DEBUG:brother_ql.reader:Byte 19              Phase type: 01
DEBUG:brother_ql.reader:Byte 20     Phase number (high): 00
DEBUG:brother_ql.reader:Byte 21      Phase number (low): 00
DEBUG:brother_ql.reader:Byte 22     Notification number: 00
DEBUG:brother_ql.reader:Byte 23                Reserved: 00
DEBUG:brother_ql.reader:Byte 24                Reserved: 00
DEBUG:brother_ql.reader:Media type: Continuous length tape
DEBUG:brother_ql.reader:Status type: Phase change
DEBUG:brother_ql.reader:Phase type: Printing state
DEBUG:brother_ql.backends.helpers:TIME 5.043 - result: {'status_type': 'Phase change', 'phase_type': 'Printing state', 'media_type': 'Continuous length tape', 'media_width': 62, 'media_length': 0, 'errors': []}
WARNING:brother_ql.backends.helpers:'printing completed' status not received.
WARNING:brother_ql.backends.helpers:'waiting to receive' status not received.
WARNING:brother_ql.backends.helpers:Printing potentially not successful?```

Any help would be much appreciated :)
Thanks!

rec0de, 2018-10-19:
The printout was about square-ish on a 62mm roll - I'm not sure if that qualifies as a long printout, but the printer did take a while so I guess a timeout issue is reasonable. 
I can't replicate that exact print size right now, but the issue seems to have disappeared in the new version. Thanks a lot!

closed 2018-10-19 by rec0de

mazz, 2021-04-14:
@rec0de > Update: I got it to work by using file:///dev/usb/lp0 as the printer URI instead of usb://0x04f9:0x2042 with pyusb backend.

Are you using macOS? I'm having a similar issue via pyusb. how did you get that `file:///` printer address? 

my issue:

```
Need to resize the image...
Traceback (most recent call last):
  File "stickers.py", line 354, in <module>
    Main()
  File "stickers.py", line 54, in __init__
    printfolder.start(args)
  File "stickers.py", line 254, in start
    self._printfolder_images(sorted_array, int(args.sticker_count), args.sticker_size)
  File "stickers.py", line 275, in _printfolder_images
    send_to_printer(random_sticker["filepath"])
  File "stickers.py", line 141, in send_to_printer
    send(print_data, PRINTER_IDENTIFIER)
  File "/Users/michael/bin/mypyenv/lib/python3.8/site-packages/brother_ql/backends/helpers.py", line 61, in send
    printer.write(instructions)
  File "/Users/michael/bin/mypyenv/lib/python3.8/site-packages/brother_ql/backends/generic.py", line 32, in write
    self._write(data)
  File "/Users/michael/bin/mypyenv/lib/python3.8/site-packages/brother_ql/backends/pyusb.py", line 149, in _write
    self.write_dev.write(data, int(self.write_timeout))
  File "/Users/michael/bin/mypyenv/lib/python3.8/site-packages/usb/core.py", line 408, in write
    return self.device.write(self, data, timeout)
  File "/Users/michael/bin/mypyenv/lib/python3.8/site-packages/usb/core.py", line 979, in write
    return fn(
  File "/Users/michael/bin/mypyenv/lib/python3.8/site-packages/usb/backend/libusb1.py", line 837, in bulk_write
    return self.__write(self.lib.libusb_bulk_transfer,
  File "/Users/michael/bin/mypyenv/lib/python3.8/site-packages/usb/backend/libusb1.py", line 938, in __write
    _check(retval)
  File "/Users/michael/bin/mypyenv/lib/python3.8/site-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])

```

rec0de, 2021-04-14:
I was using Linux at the time, where the first available printer is fairly predictably lp0, that's where I got the address from. This might not work for macOS.

Anyways it's been a few years so I'm hazy on the details. Good luck debugging!

mazz, 2021-04-15:
@rec0de thanks for the feedback. 

I found a workaround: call `os.system()` instead. I know you've probably moved-on but this message might help someone else :)

`os.system("brother_ql --model QL-800 --backend pyusb --printer usb://0x04f9:0x209b print -d --label 62 {}".format(random_sticker["filepath"]))`