pklaus/brother_ql #47: QL-820NWB Permissions/USB Error

fnlnfv, 2018-10-18:
I'm able to print labels using the Brother P-Touch Editor via USB but I'm unable to do that via brother_ql. I've tried it on both Ubuntu and Windows 7 with both USB and the Linux Kernel. 

Using the print or send command (eg. `brother_ql -b pyusb -m QL-820NWB -p usb://0x04f9:0x209d print -l 62 test.png`), I get `usb.core.USBError: [Errno None] b'libusb0-dll:err [submit_async] submitting request failed, win error: The parameter is incorrect.\r\n'`

It also happens when trying to send an instructions file. On Linux, I get a permissions denied error.

pklaus, 2018-10-19:
Hi @fnlnfv, thanks for your report.
**On Windows:**
Did you use the libusb version that I'm recommending in the README: libusb-win32-devel-filter-1.2.6.0.exe? Could you please post the full stacktrace of the exception? What you could try is to get the *Latest Windows Binaries* from <https://libusb.info/> and try with those. Please uninstall the libusb-win32... software in that case. (One more alternative: Set up the libusb support for your label printer with Zadig <https://zadig.akeo.ie/>, you may try the libusbK variant.)
**On Linux:**
Just to quickly try if you can make it work, using `sudo ...` can be used. A better fix is to set the permissions correctly. If you're using the `linux_kernel` backend on Linux, you can check the permissions of `/dev/usb/lp0` and make sure your user account is in that group. If you're also using the PyUSB backend on Linux, then check the permissions on `/dev/bus/usb/xxx/yyy`.

fnlnfv, 2018-10-23:
Windows:
I was using the libusb .exe that was recommended in the README. Using Zadig, I updated two drivers for the printer. On Interface 0, I updated the driver to libusbK (v.3.0.7.0). on Interface 1, I updated the driver to WinUSB (v6.1.7600.16385). I wasn't able to properly connect without updating either. (Wrong roll size errors)

closed 2018-10-23 by fnlnfv