pklaus/brother_ql #50: Disable auto power off

obeone, 2018-10-22:
Hi,

Is there a solution using brother_ql to disable auto power off of the printer ? 

I know it's possible using Windows or Mac utilities, but I'm working on a project which will use Brother printers at large scale and I would like to be able to disable it directly from linux, avoiding the need to pre-configure the printer.

pklaus, 2018-10-22:
Hi Grégoire,

while I think that it's not part of the raster language, I guess this
should be perfectly possible with some code based on the PyUSB backend.
Unfortunately with my current workload, it would take me some time until I
can sit down, do the required reverse engineering, programming and testing
necessary to make it work. Feel free to contact me via email (see Readme)
about sponsored development options, which could speed up things.

Best regards,
Philipp

>

obeone, 2018-10-23:
Hi Philipp,

Thanks for you reply.
Ok about that, I will see if we can do something, and which solution the project manager will choose (we are not 100% sure at this time to use Brother printer)

raat1979, 2019-01-03:
Actually raster commands for this do exist, did a usb capture under windows last night using the brother tools to change these settings of a ql-800. I still need to try some of the functions but for now I have the auto power off timeout settings figured out.  Still need to write some code to strip unneeded sequences (the tool always does the full init: esc @, status information request, switch to raster etc, etc, ) and check if it works,
don't have any python knowlege though...

raat1979, 2019-01-03:
will continue on this tonight/ tomorrow...

sako0938, 2019-03-01:
Curious about this on this QL-800. My big problem currently is once the printer auto-shuts off, once I turn it back on manually I continually get ```usb.core.USBError: [Errno 19] No such device (it may have been disconnected)``` errors. Even when unplugging the usb, and restarting the program. A reboot is required for me.

sako0938, 2019-05-14:
I was able to create a USB bus capture by installing the windows drivers, installing "Device Monitoring Studio" by HHD software, and then using the Printer Settings Tool to update the printer. 

I enabled "Auto Turn On", which so far doesn't actually turn it on once plugging it in. Also I set Auto-Poweroff to None. We'll see if it stays on indefinitely. Could be a good workaround if this setting stays somewhere in the device memory so I dont have to keep booting into Windows.

Abhi0725, 2020-02-05:
@pklaus Any chance we can get this?

meirg, 2020-06-14:
Here are the codes for setting from the command line in Linux, sniffed from QL-800:
```
# Disable Auto Power-Off
echo -n -e '\x1b\x69\x55\x41\x00\x00' | brother_ql send -
# Auto Power-Off After 10 Minutes
echo -n -e '\x1b\x69\x55\x41\x00\x01' | brother_ql send -
# Auto Power-Off After 20 Minutes
echo -n -e '\x1b\x69\x55\x41\x00\x02' | brother_ql send -
# Auto Power-Off After 30 Minutes
echo -n -e '\x1b\x69\x55\x41\x00\x03' | brother_ql send -
# Auto Power-Off After 40 Minutes
echo -n -e '\x1b\x69\x55\x41\x00\x04' | brother_ql send -
# Auto Power-Off After 50 Minutes
echo -n -e '\x1b\x69\x55\x41\x00\x05' | brother_ql send -
# Auto Power-Off After 60 Minutes
echo -n -e '\x1b\x69\x55\x41\x00\x06' | brother_ql send -
```
For auto power-on (not sure what this actually does)
```
# Enable Auto Power-On
echo -n -e '\x1b\x69\x55\x70\x00\x01' | brother_ql send -
# Disable Auto Power-On
echo -n -e '\x1b\x69\x55\x70\x00\x00' | brother_ql send -
```

raat1979, 2020-06-15:
"Auto" power on should be the state when plugging in the device

kosmic247, 2020-07-27:
@meirg Solution works for QL-700 model, too. Thank you so much :+1: 

On linux systems the codes can be also set by a simple `echo` to the device without using brother_ql send:

```
echo -n -e '\x1b\x69\x55\x41\x00\x00' > /dev/usb/lp0
```

To make this settings persistent, we used a simple udev rule that gets triggered whenever the device is added by the kernel (boot, hotplug, etc.):

```
# /etc/udev/rules.d/99-brotherql.rules

ACTION=="add", SUBSYSTEMS=="usb", KERNEL=="lp0", RUN+="/usr/bin/disable_poweroff.sh /dev/usb/lp0"
```

Custom disable power-off script
```
# /usr/bin/disable_poweroff.sh

#!/bin/bash
echo -n -e '\x1b\x69\x55\x41\x00\x00' > $1
```

Make it executable
```
chmod +x /usr/bin/disable_poweroff.sh
```

Reload the rules
```
udevadm control --reload-rules && udevadm trigger
```

caish5, 2021-02-23:
Thanks so very much.
You have no idea how many QL-700s I've temporarily install in Windows VMs
just to get rid of the 60 min poweroff.

On the subject of persistence. 
I don't actually think you need to do anything, just set it and then they've always been fine
on Raspberry Pis.