pklaus/brother_ql #52: QL-820NWB Error wrong roll type

akilinomendez, 2018-11-27:
Hi i run this command:
brother_ql  print -l 62 tmp.png

And in terminal get: 
WARNING:brother_ql.conversion:Need to resize the image...
INFO:brother_ql.backends.helpers:Sending instructions to the printer. Total: 5731 bytes.

But in printer error wrong roll type

akilinomendez, 2018-11-30:
IF  not pass de command --red or black not print. 

With roll 62 not problemn but the de rolls 17x54|17x87|23x23|29x42|29x90|39x90|39x48|52x29|62x29|62x100|102x51|102x152|d12|d24|d58
display same error.

inneralien, 2018-12-23:
I figured out the issue after reading the command line help more thoroughly. Since the sample 62mm roll that comes with the unit is a red/black the `--red` option has to be passed on the command line.

`brother_ql print -l62 --red sample_label.png
INFO:brother_ql.backends.helpers:Sending instructions to the printer. Total: 74644 bytes.
INFO:brother_ql.backends.helpers:Printing was successful. Waiting for the next job.`

bobbysmith007, 2019-12-03:
Its very easy to miss that "--red" must be supplied with certain roll sizes and models.   

I wonder if it would be possible to default to --red when it must be true (-l 62  seems to require --red on QL-810W and QL-820)  

We could probably close this ticket, but it is useful for others to find this issue.
Thanks for pointing that out @inneralien

akilinomendez, 2019-12-14:
Exactly @bobbysmith007 . It is not closed because it can be helpful.

bobbysmith007, 2020-02-03:
Is there a way to detect the roll type that the machine recognizes internally.  It would be nice to be able to insert either roll and have it detect and use the correct one.  I dont see any options for retrieving info from the printer about what is loaded