pklaus/brother_ql #54: Usage example/documentation for using in another python script instead of the CLI?

retrojim, 2018-12-21:
Is there any documentation or a usage example for using brother_ql as a module rather than from the CLI? I can call the command line from my python script just fine but I find that a little inelegant. I was hoping there might be some documentation of the functions and what to include that might save me having to trace through all the code and come up with such documentation myself.

hydroid7, 2019-01-08:
Here is a simple example. It is tested on the QL 700 printer and working fine.
```python
import brother_ql
from brother_ql.raster import BrotherQLRaster
from brother_ql.backends.helpers import send

# Using USB connected printer 
PRINTER_IDENTIFIER = '/dev/usb/lp1'

printer = BrotherQLRaster('QL-700')
def sendToPrinter(path):
    filename = 'path-to-file'
    print_data = brother_ql.brother_ql_create.convert(printer, [filename], '62', dither=True)
    send(print_data, PRINTER_IDENTIFIER)
```

retrojim, 2019-01-08:
Excellent! Thanks so much!

hydroid7, 2019-03-07:
@pklaus is it worth to add it to the README?