pklaus/brother_ql #56: model unknown QL-810W

webeasyplay, 2019-01-12:
I installed libusb-win32 and key "brother_ql -b pyusb discover" 
It's display Found a label printer: usb//0x04f9:0x209c <model:unknown>

My device model name is QL-810W

paveltyk, 2019-01-31:
I have the same issue with QL-700. I've connected two printers, and without device serial number I can't address the printer correctly. Here my output on RapsberryPI 3:

pi@raspberrypi:~ $ brother_ql -b pyusb discover
INFO:brother_ql.output_helpers:  Found a label printer: usb://0x04f9:0x2042  (model: unknown)
INFO:brother_ql.output_helpers:  Found a label printer: usb://0x04f9:0x2042  (model: unknown)
usb://0x04f9:0x2042usb://0x04f9:0x2042

paveltyk, 2019-01-31:
Also, I get this output (the printer prints image succesfully):

```
pi@raspberrypi:~ $ brother_ql -b pyusb -m QL-700 -p usb://04f9:2042/000A7Z479144 print -l 62 ./eventsme-ticket-2785346151.png 
WARNING:brother_ql.conversion:Need to resize the image...
INFO:brother_ql.backends.helpers:Sending instructions to the printer. Total: 64964 bytes.
Traceback (most recent call last):
  File "/usr/bin/brother_ql", line 11, in <module>
    sys.exit(cli())
  File "/home/pi/.local/lib/python2.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/.local/lib/python2.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/pi/.local/lib/python2.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/.local/lib/python2.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/.local/lib/python2.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/.local/lib/python2.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/.local/lib/python2.7/site-packages/brother_ql/cli.py", line 147, in print_cmd
    send(instructions=instructions, printer_identifier=printer, backend_identifier=backend, blocking=True)
  File "/home/pi/.local/lib/python2.7/site-packages/brother_ql/backends/helpers.py", line 76, in send
    result = interpret_response(data)
  File "/home/pi/.local/lib/python2.7/site-packages/brother_ql/reader.py", line 159, in interpret_response
    raise NameError("Printer response doesn't start with the usual header (80:20:42)", hex_format(data))
NameError: ("Printer response doesn't start with the usual header (80:20:42)", '61 72 72 61 79 28 27 42 27 2C 20 5B 31 32 38 2C 20 33 32 2C 20 36 36 2C 20 35 32 2C 20 35 33 2C 20 34 38 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 36 32 2C 20 31 30 2C 20 30 2C 20 30 2C 20 32 31 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 5D 29')
```

arnotixe, 2021-05-05:
Hi there, not sure if it is related, but I have a similar experience with brother_ql from master and a QL-810W nordic printer on raspberry. Python 3. (Exact serial / type number E77136E0Z365358 = QL810W Fabrication date 2020 I think)

Printing is perfect including the cut/no cut option, seems only the "finished/OK" response from the printer is bad. Printer also does not hang or anything; I can print 50 stickers perfectly. Is a bit slow though because it seems to timeout after print. 10 secs maybe?

The printui fork printing whatever also says "Could not connect to printer" (but does print just fine)

Is there something I can contribute to fix? debug mode to see what the printer actually responds, or something? Probably a newer printer/firmware responding something else than what is expected I guess…

[image]