pklaus/brother_ql #57: suppress warnings

x4FF3, 2019-02-11:
Hi,

is there a way to suppress following messages:

> Trying to switch the operating mode on a printer that doesn't support the command.
> Trying to switch the operating mode on a printer that doesn't support the command.
> Need to resize the image...

thanks

waltersdmh, 2019-06-07:
Hi,

Did you manage to fix this?

My script hangs on

> brother_ql.conversion:Need to resize the image

x4FF3, 2019-06-07:
```python
import logging
log = logging.getLogger('brother_ql')
log.setLevel(50)
```

waltersdmh, 2019-06-07:
Thanks for this

I'm not sure this will work for me though. I'm using python's os.system(cmd) to run command line code that executes the brother commands. 

I dont know how to control this os.system yet. WOrking on it.