pklaus/brother_ql #58: QL-800 red blinking light when print issued

jmcwhirter, 2019-02-18:
- Model: Brother QL-800
- Hardware: Raspberry Pi B+
- OS: Raspbian GNU/Linux 9
- Connection method: USB
- The [logo](https://user-images.githubusercontent.com/527471/52926078-ed3db300-32f9-11e9-80a7-df0a0af4e61e.jpg) I want to print, as a jpg:
[image]

## Setup...
1. Tried with Python 2 and 3: `pip3 install --upgrade brother_ql`
2. Add to path: `export PATH="${PATH}:~/.local/bin"`
3. Find my printer details: `sudo lsusb -v`
4. Constructed my PyUSB string like so: usb://0x04f9:0x209b/____

## Steps I took...

1. Create bin file
``` bash
pi@raspberrypi:~ $ brother_ql_create --model "QL-800" logo.jpg --label-size 62 > logo.bin
```
2. Use bin file to print
``` bash
pi@raspberrypi:~ $ brother_ql_print logo.bin usb://0x04f9:0x209b/____
deprecation warning: brother_ql.devicedependent is deprecated and will be removed in a future release
INFO:brother_ql.backends.helpers:Sending instructions to the printer. Total: 64972 bytes.
WARNING:brother_ql.backends.helpers:'printing completed' status not received.
WARNING:brother_ql.backends.helpers:'waiting to receive' status not received.
WARNING:brother_ql.backends.helpers:Printing potentially not successful?
```
3. **red light blinks...**
4. Try to run analyse to simulate what the printer is going to get and render:
``` bash
pi@raspberrypi:~ $ brother_ql_analyse logo.bin --loglevel DEBUG
```
5. Reviewed [output](https://github.com/pklaus/brother_ql/files/2873935/analyse.log) from analyse execution and things look good.
6. I even get a [logo](https://user-images.githubusercontent.com/527471/52926077-eca51c80-32f9-11e9-840e-0ec36fc0253d.png) as a result of the analyse process, as a png
7. I try to run the debug command to see if there is any extra information.
``` bash
pi@raspberrypi:~ $ brother_ql_debug logo.bin usb://0x04f9:0x209b/____
deprecation warning: brother_ql.devicedependent is deprecated and will be removed in a future release
INFO: CMD mode setting FOUND. Instruction: 1B 69 61 01 
INFO: CMD preamble FOUND. Instruction: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [...] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
INFO: CMD init FOUND. Instruction: 1B 40 
INFO: CMD mode setting FOUND. Instruction: 1B 69 61 01 
INFO: CMD status request FOUND. Instruction: 1B 69 53 
INFO: Response from the device: 80 20 42 34 38 30 00 00 00 00 3E 0A 00 00 23 00 00 00 00 00 00 00 00 00 00 81 00 00 00 00 00 00
INFO: Interpretation of the response: 'Reply to status request' (phase: Waiting to receive), 'Continuous length tape' 62x0 mm^2, errors: []
INFO: CMD media/quality FOUND. Instruction: 1B 69 7A CE 0A 3E 00 B8 02 00 00 00 00 
INFO: CMD various FOUND. Instruction: 1B 69 4D 40 
INFO: CMD cut-every FOUND. Instruction: 1B 69 41 01 
INFO: CMD expanded FOUND. Instruction: 1B 69 4B 08 
INFO: CMD margins FOUND. Instruction: 1B 69 64 23 00 
INFO: CMD raster FOUND. Instruction: 67 00 5A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [...] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
INFO: Response from the device: 80 20 42 34 38 30 00 00 00 00 3E 0A 00 00 23 00 00 00 06 01 00 00 00 00 00 81 00 00 00 00 00 00
INFO: Interpretation of the response: 'Phase change' (phase: Printing state), 'Continuous length tape' 62x0 mm^2, errors: []
INFO: CMD print FOUND. Instruction: 1A 
INFO: Response from the device: 80 20 42 34 38 30 00 00 00 00 3E 0A 00 00 23 00 00 00 02 01 00 00 00 00 00 81 00 00 00 00 00 24
INFO: Interpretation of the response: 'Error occurred' (phase: Printing state), 'Continuous length tape' 62x0 mm^2, errors: []
```
9. Everything looks good until then end when there is an error in the printing phase that does not get bubbled up.

jmcwhirter, 2019-02-18:
Update:
1. I moved the printer over to my Mac (Sierra)
2. I tried to send the bin file via `lp logo.bin` from my Mac. Red light flashed and no print. - FAIL
3. I installed libusb and fontconfig via brew
4. I installed [Brother_QL Web](https://github.com/pklaus/brother_ql_web), but ran into a problem with the font folder. I filed an issue. - FAIL
5. I installed [Brother_QL API](https://github.com/pklaus/label_api) and this worked. - AWESOME!
6. I tried to issue a print command and the red light flashed again. - FAIL

sako0938, 2019-02-18:
I am currently just getting started with using this library with the QL-800. Currently working through the code running on a Pi 3B+, but can do some basic prints using code similar to this:
```
import brother_ql
from brother_ql.raster import BrotherQLRaster
from brother_ql.backends.helpers import send, discover

# Using USB connected printer 
PRINTER_IDENTIFIER = '/dev/usb/lp0'

printer = BrotherQLRaster('QL-800')
filename = "/home/pi/Documents/test.png"
print_data = brother_ql.brother_ql_create.convert(printer, [filename], '39x90', dither=True)
send(print_data, PRINTER_IDENTIFIER)
```
Make sure the "Editor Lite" button is not active, for me it won't show up on USB with that on. Maybe we can help each other with this printer!

clee505, 2019-02-20:
I'm on Ubuntu using Python3. Running into the same issue but when I open top and close, it does not print. Running a 17x87 label and testing with resized images. The error i'm getting is "Replace media error"

sako0938, 2019-02-21:
I have gotten prints working on my Ubuntu PC, and on Raspberry Pi with this code. So far I need to re-initialize the printer object before every print:
```
import brother_ql
from brother_ql.conversion import convert
from brother_ql.raster import BrotherQLRaster
from brother_ql.backends.helpers import send, discover

# Pyusb backend
PRINTER_IDENTIFIER = discover('pyusb')[0]['instance']
# Linux Backend
# PRINTER_IDENTIFIER = '/dev/usb/lp0'
# Need to create new qlr object for each print
qlr = BrotherQLRaster("QL-800")
qr_data = convert(qlr,["/home/youruser/Desktop/yourpic.png"],"29x90")
send(qr_data, PRINTER_IDENTIFIER)
```

jmcwhirter, 2019-02-21:
Interesting stuff! I was being too stubborn and didn't try different sized paper or the Python library. I'll have to give this a shot. Thanks for the info!!!

mattytrentini, 2019-04-08:
Looks like you've got this sorted but I've noticed that if I try to print from P-Touch and have selected a different size print to the loaded label the same error - flashing red light - occurs.

alistat, 2019-07-12:
Am late on this but the provided 62mm continuous tape for the QL-800 is a reb/black tape.
As the command's help states, if you are printing on a red/black tape you must pass the `--red` flag:

```
--red                 Create a label to be printed on black/red/white tape
                        (only with QL-8xx series on DK-22251 labels). You must
                        use this option when printing on black/red tape, even
                        when not printing red.
```

So if you are using the 62mm tape that was included in the package, this flag will solve your problem.

ashlynblack, 2022-03-30:
I had the red blinking power light, too. I was specifying the wrong label size!