pklaus/brother_ql #62: Printing in "pink" not supported on QL-810W

gusnasis, 2019-04-07:
With the Brother P-touch software I can add a photo on a label and it'll print in pink, like this:

[image]

Trying to print the same thing through the command line will print something like this:

[image]

Is it possible for the library to support the first style of printing? It seems the printer is capable of it.

BadiuAlexandru, 2019-04-07:
Hello @gusnasis, can you share the code with me? I cannot make my QL-810W printer work with this plugin and I am really newb to python world

gusnasis, 2019-04-07:
I used the command line tool. Something like:

`brother_ql print --red -l 62 -d label.png`

(in this case using the special black/red 62mm continuous roll labels)

For this to work you need to first set some environment variables, depending on how your printer is connected. For me it was something like:

```
export BROTHER_QL_MODEL=QL-810W
export BROTHER_QL_PRINTER=tcp://10.0.1.5
```

EilersT, 2019-11-19:
Dithering is not implemented in Red and Black mode.

gusnasis, 2019-11-19:
Would it be possible to implement it? Any thoughts on how to go about it? It seems the hardware supports it since the Brother software will print this way.

EilersT, 2019-11-19:
[image]

I'm trying to code it in but it will take some time. I'm an absolute NOOB to Python and I'm currently just figuring out Git & GitHub. I will post if its going somewhere. No promises!