pklaus/brother_ql #63: Printer prints multiple times if Python interface is used.

der1igedon, 2019-04-10:
Hello,

first of all thank you very much for your work!
I found a bug using the Python interface for the printer.

The following code prints 3 times with only 2 times calling the print command:

```
import brother_ql
from brother_ql.raster import BrotherQLRaster
from brother_ql.backends.helpers import send

# Using USB connected printer
PRINTER_IDENTIFIER = 'tcp://131.246.248.199:9100'

printer = BrotherQLRaster('QL-720NW')

def sendToPrinter():
    filename = 'name1.png'
    print_data = brother_ql.brother_ql_create.convert(printer, [filename], '62', dither=True)
    send(print_data, PRINTER_IDENTIFIER)

sendToPrinter()
sendToPrinter()
```

baileyds, 2019-06-07:
I'm seeing some similar oddities using the network connected version of a brother printer - it will print multiples of the same item, though not consistently.  Possibly a code issue but I had to comment seeing as this seemed so similar.  Will update with a code example if helpful.

mattj23, 2019-09-18:
I had this issue too, I ended up solving it by creating a new raster between calls to print and not re-using it.  

The issue that I was seeing in particular was that the raster would print every single image that it had been given:

I sent Image A => printer printed A

I sent Image B => printer printed B, then A

I sent Image C => printer printed C, then B, then A

After I started deleting the `BrotherQLRaster` and creating a new one for each send, the problem went away.

michalmph, 2020-05-22:
Can confirm, I'm having this issue with current master as well. 

I'm printing labels from a loop, and I need to 
`printer = BrotherQLRaster('QL-1050')`
after every print call to the printer, otherwise it prints all the previous labels.