pklaus/brother_ql #64: pillow dependency issue

andrematzke, 2019-04-14:
for installing of brother_ql  
pacman based
I got always a problem when installing. It tries always something I do not understand.

`Searching for pillow>=3.3.0
Reading https://pypi.python.org/simple/pillow/
Downloading
https://files.pythonhosted.org/packages/81/1a/6b2971adc1bca55b9a53ed1efa372acff7e8b9913982a396f3fa046efaf8/Pillow-6.0.0.tar.gz#sha256=809c0a2ce9032cbcd7b5313f71af4bdc5c8c771cb86eb7559afd954cab82ebb5
Best match: Pillow 6.0.0`

why is the best match Pillow 6.0.0 ? 
With pacman I have installed a system-wide Pillow 5.2 (if it comes to system-wide, one should always install with pacman, not with pip). Since your setup.py demands Pillow >3.3 it should meet the necessary requirements.
Why the 6.0.0 ?

on pacman systems your software cannot be installed. 6.0.0 is not yet in standard arch repos.

Best Regards,
Andre

radarhere, 2019-04-24:
https://pip.pypa.io/en/stable/reference/pip_install/
> Once pip has the set of requirements to satisfy, it chooses which version of each requirement to install using the simple rule that the latest version that satisfies the given constraints will be installed

madeinaus, 2023-03-23:
On Windows XP/python 2.7.18, pip installs pillow 6.x, even though that is not compatible with WinXP. I think that it is because 6.x is the last available 32-bit  version