pklaus/brother_ql #67: Has anybody tried this with P series brother printers?

lukio3, 2019-08-27:
Thanks so much for open sourcing this.

They don't appear in the supported list, but i was wondering if any of the logic herein would or could apply to brother [p-series ](www.brother.co.uk/labelling/ptouch-labellers/ptp900w) printers?

We have al QL series printer at work too, but the P series is the one used for smaller labels, and i'm trying to find a solution that would be compatible with P series and QL series.

Luke

lazyzero, 2019-08-31:
I have a P700 but I am struggling with getting it to work with Linux. I found this repository: 
https://git.scc.kit.edu/scc-net/labelprinterkit
My python skills are only measurable on a negative scale (I was unable to use their example at all).
Maybe it helps @pklaus to add P700 support to brother_ql?

I would be more than happy to help testing.

Christian

pklaus, 2019-09-05:
Thanks @lukio3 and @lazyzero for your ideas and hints. Yes, actually I own a P-touch label printer and I wanted to add support for quite some time. The larger commit 0a3af68 was a big step towards supporting additional models and labels. Now 05516a7 added support to analyze binary instruction files for the P-touch series and 638b365 a first draft to support printing on the PT-P750W. What's a bit annoying is that the P-touch series label printers do not all have the same resolution. So the number of printable area for example for the 24mm tape is 128 pixels wide on the PT-P750W and 320 pixels wide on the PT-900W. Currently, my data structure for the labels doesn't allow to specify different pixel widths for different models. So some more thought has to be put into this. In the meantime I'll have a closer look at labelprinterkit, too.
I'll be busy over the weekend but next week there should be something to test.

lukio3, 2019-09-06:
@pklaus thanks so much. Looking forwarding to hearing where you get up to.

dondelelcaro, 2019-09-16:
@pklaus I'm also interested in the PT-2430PC, so happy to try to test/submit patches for that once things seem to be working reasonably on some other PT models. [Don't really have enough time currently to contribute much more than that, but excited to eventually get your web interface working with my label printer. ;-)]

ristomatti, 2019-10-03:
@pklaus Please also consider looking into PT-P300BT? :) I ran across this GIST https://gist.github.com/dogtopus/64ae743825e42f2bb8ec79cea7ad2057 which provides a CLI tool that connects to the printer using pybluez and supports printing 128px wide PNG's.

NotAFile, 2020-02-22:
Hi, I'm the author of Labelprinterkit. Someone just pointed me to this issue. I thought I'd give a few thoughts.

 1. The brother docs are incorrect. They state a `'g'` character for line transfer, instead of `'G'`. This is already correct in this repo, I thought I'd just mention it for any other unfortunate souls passing through.
 2. I really wish there had already been this much freely available source code when I struggled with the printer two years ago... How did apparently everyone figure out the docs being incorrect but me?
 3. I remember the PackBits compression being very important, which `brother_ql` does not support. I think I ran into some semi-documented size limit, and significant increases in printing time.
 4. I don't remember why I didn't add support for my printer to `brother_ql` instead of building my own driver into `labelprinterkit`. I think it was a combination of it being unsupported, being unsure how much work it would be, concerns about code quality and worries about relying on what seemed like an undocumented and unofficial internal API of a cli tool.

Either way, looking at it currently, I think it would be relatively easy to move the code over. It seems wasteful to further duplicate device driver functionality in `labelprinterkit` to add things like networking and bluetooth. However, my code works for what I need it for, so it's unfortunately not a big priority for me.

lsr1, 2020-04-22:
Hi, thank you so much for this code! Do you  think it would make sense / is possible to integrate the project probonopd/ptouch-770 here? This would be awesome - b/c then the awesome  pklaus/brother_ql_web could be used to print labels on a p770 via web \o/

Pararera, 2020-07-01:
Hi @pklaus , sorry for refresh.
Do you still work on adding P series to this library?

scottepeterson, 2020-10-23:
I'm interested in PTP900W and PTP700 support.  I see the 900 and the 750 listed in models.  Anybody know if this is still 'beta' or if its at all solid (don't have them on my desk here to test).   If not I might be willing to give it a shot to add it in - but not sure what level of a project that is... any thoughts as to the level of effort to add a model?

snicker, 2021-02-13:
Also interested in assisting support for the PT2430 printer, but I don't have the slightest clue where to start. This C library communicates directly over USB and seems to have most of the protocol understood:
https://github.com/clarkewd/ptouch-print
Not sure if that available information is enough to start porting to this python library

kettenbach-it, 2021-04-08:
Thanks so much for open sourcing this.
I would be interested in support for PT-P750W

NotAFile, 2021-04-08:
@kettenbach-it I have one next to my desk right now, but I haven't worked on it yet. As far as I can tell, the protocol is identical except for the submission over IP.

kettenbach-it, 2021-04-08:
I'm using both my devices PT-P750W and QL-720NW via IP

Ravens28, 2023-08-13:
Hi,
Has someone tried modifying the great job of pklaus to adapt it with the P-series. I can try it with my p300bt.
I'm willing to try to adapt the code myself but I have to admit that I'm not sure to have the proper skills 😅