pklaus/brother_ql #68: Run it on client's machine

Abhi0725, 2019-09-09:
We have a software that prints labels. Unlike Dymo printers I don't find any JS SDK for Brother printers. My question is how my customers will be able to get the tags printed when they are only on our web app?

My guess is we can create an electron app that our customers will have to install which will install python in their system along with the brother_ql package. Then that electron app will use websockets to keep a connection with the server which will keep sending print jobs. Is this the correct way to do? Do have any electron app that is already built for this purpose?

Abhi0725, 2019-09-12:
@pklaus can you help in this?

pklaus, 2019-09-14:
Dear @Abhi0725 thanks for your interest in the software. I do not fully get your full architecture.

As far as I know, nobody has built an electron application with Python and brother_ql embedded so far.

I guess, the end users of your software will be sitting in front of a Windows PC that should run a GUI to design or create the labels to be printed (because you mention a possible electron app). On the other hand, you write you're customers are on a "web app". So which of those is the case then?

In any case, I want to point you to two projects that are built on top of brother_ql, where printing via HTTP requests is possible:
https://github.com/pklaus/label_api
https://github.com/pklaus/brother_ql_web
You could call this a "JS SDK" in a wider sense. Off course, this software and the two derived projects above are in Python, so the server side will be in Python. On the client you can use Javascript or any other technology able to emit HTTP requests.

Please let me know more about your use case so that I can give you a better recommendation.

Abhi0725, 2019-09-15:
Thanks for your detailed reply @pklaus and sorry if my question was not clear.
Actually my case is we have a check-in system. Every client will have our web app opened on their browser (which is written in PHP). Whenever they hit check-in button the tags needs to be printed on the client side. The client can have a brother printer attached to the same computer or on a different computer. Also these check-in can also happen via mobile app hence all the print jobs will need to go to server.

Also, as this `brother_ql` package is written on python so I suppose the client will need to have python installed on their system (which is not possible as the clients are not tech-savvy).

Hence I decided to build an electron app which, when a client will download and install, will install python in their system and the electron app will keep a socket connection open to the server to receive print jobs.

So in a nutshell the customers/clients needs just install our software and plug the printer they shouldn't know that they need to install python on their system.

Is this the right approach? Let me know if you need more details.

Also, you said that the server side will be in python and client just need to emit the HTTP request. But how then the server will print labels on client's local machine? 🤔 I am missing something here?