pklaus/brother_ql #69: Printer respond doesn't start with the usual header

mapadj, 2019-10-20:
Hello.

I am trying to print my own images with the Label printer, but i am failing. It is only usable by P-Touch so far.

I'm getting the following Error Message:

`mac:labelprinter marcel$ brother_ql -m QL-800 -b pyusb discover
INFO:brother_ql.output_helpers:  Found a label printer: usb://0x04f9:0x209b  (model: unknown)
usb://0x04f9:0x209b
mac:labelprinter marcel$ brother_ql -m QL-800 -b pyusb -p usb://0x04f9:0x209b print -l 62 label.png
INFO:brother_ql.backends.helpers:Sending instructions to the printer. Total: 25447 bytes.
Traceback (most recent call last):
  File "/usr/local/bin/brother_ql", line 10, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 696, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 1060, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/brother_ql/cli.py", line 147, in print_cmd
    send(instructions=instructions, printer_identifier=printer, backend_identifier=backend, blocking=True)
  File "/usr/local/lib/python2.7/site-packages/brother_ql/backends/helpers.py", line 76, in send
    result = interpret_response(data)
  File "/usr/local/lib/python2.7/site-packages/brother_ql/reader.py", line 164, in interpret_response
    raise NameError("Printer response doesn't start with the usual header (80:20:42)", hex_format(data))
NameError: ("Printer response doesn't start with the usual header (80:20:42)", '61 72 72 61 79 28 27 42 27 2C 20 5B 31 32 38 2C 20 33 32 2C 20 36 36 2C 20 35 32 2C 20 35 36 2C 20 34 38 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 36 32 2C 20 31 30 2C 20 30 2C 20 30 2C 20 33 35 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 31 32 39 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 5D 29')
mac:labelprinter marcel$`

Also Printer starts to blink red, after print attempt.

I am using this picture:
[image]

I have the feeling that the file format might be wrong. Since i am also not able to use my images with the standard print function of macos. It might be connected to additional borders, but it doensn't print smaller images as well.

I have a 62mm black/red tape endless tape. I have to select exactly this option in p-touch to make it work. But i don't see this option in brother_ql. I tried all of them 62's anyways. Neiter worked out for me.
 
Can anyone help me learn, what i am doing wrong?

jonaswiberg, 2019-11-18:
@mapadj, did you solve your issue?  I'm facing the same :(

mapadj, 2019-11-19:
@jonaswiberg, unfortunately i did not so far. I will give it a new try next week. I will keep you posted. But right now it looks like i will switch to zebra printers.

jonaswiberg, 2019-11-20:
@mapadj, ok - too bad, I'm also stuck. Appreciate that you keep me posted. If I make any progress I will definetely let you know.

pklaus, 2019-12-03:
Hello @mapadj, hello @jonaswiberg,

1st of all, thanks for reporting this issue here on Github. You found a proper bug.

2nd, I found the following. The printer response contains a lot of bytes in hex notation. Usually this should be 32 bytes but in your case it's a lot more. When I looked at the data, I thought this could be some ASCII string and indeed:

```
(py37) [pklaus@jay Desktop]$ python
Python 3.8.0 (default, Oct 23 2019, 18:51:26) 
[GCC 9.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> s = """61 72 72 61 79 28 27 42 27 2C 20 5B 31 32 38 2C 20 33 32 2C 20 36 36 2C
... 20 35 32 2C 20 35 36 2C 20 34 38 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 36
... 32 2C 20 31 30 2C 20 30 2C 20 30 2C 20 33 35 2C 20 30 2C 20 30 2C 20 30 2C 20
... 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 31 32 39 2C 20
... 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 5D 29"""
>>> 
>>> b = bytes.fromhex(s)
>>> 
>>> print(b.decode('ascii'))
array('B', [128, 32, 66, 52, 56, 48, 0, 0, 0, 0, 62, 10, 0, 0, 35, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 129, 0, 0, 0, 0, 0, 0])
```
What you get back is not the printer's response but a string version of a python `array()`. This is definitely some bug in my code.

Let's see what this ASCII-encoded array string contains:

```
>>> d = bytes([128, 32, 66, 52, 56, 48, 0, 0, 0, 0, 62, 10, 0, 0, 35, 0, 0, 0, 0, 0,
... 0, 0, 0, 0, 0, 129, 0, 0, 0, 0, 0, 0])
>>> 
>>> print(' '.join('%02X' % byte for byte in d))
80 20 42 34 38 30 00 00 00 00 3E 0A 00 00 23 00 00 00 00 00 00 00 00 00 00 81 00 00 00 00 00 00
```

This looks more like a regular Brother QL-series printer response. What's in that message?

```
>>> import brother_ql.reader
deprecation warning: brother_ql.devicedependent is deprecated and will be removed in a future release
>>> brother_ql.reader.interpret_response(d)
{'status_type': 'Reply to status request', 'phase_type': 'Waiting to receive', 'media_type': 'Continuous length tape', 'media_width': 62, 'media_length': 0, 'errors': []}
```

That's basically a regular message that the printer is waiting to receive instructions and currently is loaded with a 62mm continuous label roll.

This is not yet the error message itself, which will arrive some moments later. But the incorrectly formatted response causes the software to halt already at this point.

So what's the solution for your problem? Wait for the next comment just below...

pklaus, 2019-12-03:
To solve your problem printing on the Black/Red tape (also the sample roll that comes with the QL-800 series printers):

Use the `--red` option on the command line. To cite the CLI help:

```
  --red                           Create a label to be printed on
                                  black/red/white tape (only with QL-8xx
                                  series on DK-22251 labels). You must use
                                  this option when printing on black/red tape,
                                  even when not printing red.
```

jtrexr2, 2019-12-24:
Hello @pklaus, thanks for calling out the `--red` option. I was able to use that to get my QL-800 printing from Python on the sample 62mm roll that comes with the printer. However, my printer also came with a 29mm endless roll, and I'm having the same problem as above. I've tried with and without the `--red` option, with the same `NameError` result:

```
brother_ql --backend pyusb --model QL-800 --printer usb://0x04f9:0x209b  print --red -l 29 qr.png
WARNING:brother_ql.conversion:Need to resize the image...
INFO:brother_ql.backends.helpers:Sending instructions to the printer. Total: 57160 bytes.
Traceback (most recent call last):
  File "C:\Python27\Scripts\brother_ql-script.py", line 11, in <module>
    load_entry_point('brother-ql==0.9.dev0', 'console_scripts', 'brother_ql')()
  File "c:\python27\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\python27\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\python27\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\python27\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\python27\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\python27\lib\site-packages\click\decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\python27\lib\site-packages\brother_ql\cli.py", line 147, in print_cmd
    send(instructions=instructions, printer_identifier=printer, backend_identifier=backend, blocking=True)
  File "c:\python27\lib\site-packages\brother_ql\backends\helpers.py", line 76, in send
    result = interpret_response(data)
  File "c:\python27\lib\site-packages\brother_ql\reader.py", line 164, in interpret_response
    raise NameError("Printer response doesn't start with the usual header (80:20:42)", hex_format(data))
NameError: ("Printer response doesn't start with the usual header (80:20:42)", '61 72 72 61 79 28 27 42 27 2C 20 5B 31 32 38 2C 20 33 32 2C 20 36 36 2C 20 35 32 2C 20 35 36 2C 20 34 38 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 32 39 2C 20 31 31 2C 20 30 2C 20 30 2C 20 31 2C 20 30 2C 20 30 2C 20 39 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 31 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 5D 29')
```

Any ideas here?

rickvipond, 2020-01-20:
Hi I get this error too but the --red flag is not an option for me as I have a QL500. See:

`brother_ql --printer usb://0x04f9:0x2015 --model QL-500 print -l 62 drawn_image.png
INFO:brother_ql.backends.helpers:Sending instructions to the printer. Total: 37424 bytes.
Traceback (most recent call last):
  File "/home/pi/.venvs/pos27b/bin/brother_ql", line 8, in <module>
    sys.exit(cli())
  File "/home/pi/.venvs/pos27b/local/lib/python2.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/.venvs/pos27b/local/lib/python2.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/pi/.venvs/pos27b/local/lib/python2.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/.venvs/pos27b/local/lib/python2.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/.venvs/pos27b/local/lib/python2.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/.venvs/pos27b/local/lib/python2.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/.venvs/pos27b/local/lib/python2.7/site-packages/brother_ql/cli.py", line 147, in print_cmd
    send(instructions=instructions, printer_identifier=printer, backend_identifier=backend, blocking=True)
  File "/home/pi/.venvs/pos27b/local/lib/python2.7/site-packages/brother_ql/backends/helpers.py", line 76, in send
    result = interpret_response(data)
  File "/home/pi/.venvs/pos27b/local/lib/python2.7/site-packages/brother_ql/reader.py", line 164, in interpret_response
    raise NameError("Printer response doesn't start with the usual header (80:20:42)", hex_format(data))
NameError: ("Printer response doesn't start with the usual header (80:20:42)", '61 72 72 61 79 28 27 42 27 2C 20 5B 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 31 32 38 2C 20 33 32 2C 20 36 36 2C 20 34 38 2C 20 37 39 2C 20 34 38 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 36 32 2C 20 31 30 2C 20 30 2C 20 30 2C 20 32 31 2C 20 30 5D 29')`

adding the --red flag and I get:

`raise BrotherQLUnsupportedCmd('Printing in red is not supported with the selected model.')
brother_ql.exceptions.BrotherQLUnsupportedCmd: Printing in red is not supported with the selected model.`

What is the actual fix for this?

jtrexr2, 2020-01-20:
@rickvipond Are you definitely meaning to use the 62mm endless roll? Or could you potentially be using the 62x29 or 62x100 rolls? I had problems when supplying the wrong parameter there.

pklaus, 2020-02-04:
So, I finally fixed this issue:
"Printer response doesn't start with the usual header"

I'll push the commit that closes this issue in a moment. If you still encounter a problem while printing, it's hopefully not connected to this issue. So please open another issue, for non-related issues.

commit 56cf4394 (2020-02-04): [fix] "response doesn't start with the usual header"

fixes #25, fixes #69, fixes #75, fixes #81

also seen in issues: #56, #71
closed 2020-02-04 by pklaus