pklaus/brother_ql #71: No Grayscale only B&W

EilersT, 2019-11-09:
### My Setup
- Printer: QL-800
- Tape 62mm Black/Red on White DK-22251
- OS: Ubuntu 18.04.3 LTS
- Connection USB
- Testpattern TestRG.png 696x108 px

[image]

- Printer output

[image]

### My Problem

i would like to print Pictures and Graphics with my QL-800. Unfortunately the printer produces only B&W Patterns and no gray- redscale printouts. Dithering has no effect. 
Could be related to Issue https://github.com/pklaus/brother_ql/issues/62

Any help would be much appreciated 

### Console
- Printing command:
`$ sudo brother_ql -m QL-800 -p usb://0x04f9:0x209b print -l 62red --red TestRG.png `

- Printing command with dithering
`$ sudo brother_ql -m QL-800 -p usb://0x04f9:0x209b print -l 62red -d --red TestRG.png` 

- Console output
```
$ sudo brother_ql -m QL-800 -p usb://0x04f9:0x209b print -l 62red -d --red TestRG.png 
INFO:brother_ql.backends.helpers:Sending instructions to the printer. Total: 20332 bytes.
Traceback (most recent call last):
  File "/usr/local/bin/brother_ql", line 11, in <module>
    load_entry_point('brother-ql==0.9.dev0', 'console_scripts', 'brother_ql')()
  File "/home/tilman/.local/lib/python2.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/tilman/.local/lib/python2.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/tilman/.local/lib/python2.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/tilman/.local/lib/python2.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/tilman/.local/lib/python2.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/tilman/.local/lib/python2.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "build/bdist.linux-x86_64/egg/brother_ql/cli.py", line 147, in print_cmd
  File "build/bdist.linux-x86_64/egg/brother_ql/backends/helpers.py", line 76, in send
  File "build/bdist.linux-x86_64/egg/brother_ql/reader.py", line 164, in interpret_response
NameError: ("Printer response doesn't start with the usual header (80:20:42)", '61 72 72 61 79
 28 27 42 27 2C 20 5B 31 32 38 2C 20 33 32 2C 20 36 36 2C 20 35 32 2C 20 35 36 2C 20 34 
38 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 36 32 2C 20 31 30 2C 20 30 2C 20 30 2C 20 
33 35 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 
2C 20 30 2C 20 31 32 39 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 5D 29')
```

EilersT, 2019-11-19:
Figured it out myself. Ditherting is not implemented with Red and Black printing.

pklaus, 2020-01-30:
Thanks for reporting this. Actually, it would be great if this works.
I'll see, what I can do here. Do dithering before or after deciding/thresholding between black/red pixels?
I have to try and see. For now, I'll reopen the issue.

EilersT, 2020-02-04:
I worked on the "conversion.py" that is posted on my GitHub page. Dithering is working but i cant get the contrast right.

[image]