pklaus/brother_ql #72: 23x23 (Possible other square Die-Cut sizes as well) actually capable of printing larger rectangles

PercyJax, 2019-12-04:
Using the p-touch software to print an all-black label on the 23mmx23mm square labels creates a slightly rectangular printing area. Doing the same with a 202x202 px black image prints a smaller square area.

PercyJax, 2019-12-04:
[image]

PercyJax, 2019-12-04:
Tested with
`Label("23x23",  ( 23,  23), FormFactor.DIE_CUT,       ( 272,  272), ( 250,  225),  55 ),`

at least for these labels that seems to be the sweet spot
https://www.amazon.com/Aegis-Adhesives-Compatible-Replacement-Printers/dp/B07PFRR9KH/ref=sr_1_5?keywords=dk1221&qid=1575485069&sr=8-5

PercyJax, 2019-12-04:
Submitted pull https://github.com/pklaus/brother_ql/pull/73

Mikhail5555, 2020-01-21:
I think the only way to fix this is enable local overrides, all labels/printers have just slightly different margins, your change is only 'perfect' for your printer + label combinations.
I believe the current values are so it will work for MOST printers. See my comment in your PR.