pklaus/brother_ql #74: Feature request: cut only

ZdenekM, 2020-01-06:
Would it be possible to add command to perform only cut? Like if I print label without cutting and then need to cut it later.

Mikhail5555, 2020-01-15:
You could theoretically feed it a full white image, or use the cut button on your label printer?

ZdenekM, 2020-01-17:
I need to pick the printed label with a robot (one arm only) so pressing a cut button is not an option unfortunately :-) I tried to print full white image with 1px length but probably due to some margins or whatever, the printer moves out approximately 2 cm of label and then cuts. If I print and cut, the printed label is 'fired' from the printer so I need to print it, grasp using a robot's suction cup and then cut it.

Mikhail5555, 2020-01-17:
You could possibly try removing the line that adds the magin to the printing payload (https://github.com/pklaus/brother_ql/blob/master/brother_ql/conversion.py#L188), that might lead to it 'printing' only the 1px and then cutting it.

Mikhail5555, 2020-01-18:
I did quite a bit of googling, since this was quite an interesting suggestion, sadly I didn't get a very great result.
TLDR version: This library uses Raster mode to print it on the printers. They also have an ESC/P mode (which only some printers support from my research) which does state that there is a printer operation to 'Only cut'.  (See screenshot below, taken from the developer documentations of the QL-720NW printer), So maybe just maybe it would work, but even then, it look like it still feeds a minimum amount before it cuts.

[image]

ZdenekM, 2020-01-22:
@Mikhail5555 Thank you for your effort. As a first attempt, I will try to remove margins.

N07070, 2026-11-13:
I would also be interested in such an option