pklaus/brother_ql #78: PT-P750W requires compression to be used

jpieper, 2020-01-26:
I was able to use brother_ql to print labels to a PT-P750W, but only when compression was enabled.  Without compression, it just prints blank.  Perhaps that should be a mandatory setting for that printer (and perhaps all P-Touch ones)?

raddessi, 2020-02-21:
This is also true for my very similar PT-E550W which I have been trying to get to work (and just did thanks to this comment @jpieper, thank you very much!). It also seems that setting `compression=True` in the models.py definition of the printer is not respected, I was forced to add the cli argument `compress` before I would see any text on the printed label among other issues.

bodegaard, 2021-07-06:
With --compress worked fine with a .bmp image on a PT-P750W. It also printed the image without this but it was skewed to one side. note: I was using the code from commit 56cf4394ad750346c6b664821ccd7489ec140dae and not the released version.