pklaus/brother_ql #81: QL800: Traceback when printing PNG file from Raspberry Pi

henningwerner, 2020-01-27:
```
brother_ql print -l 62x100 example-label.png
No handlers could be found for logger "brother_ql.devicedependent"
INFO:brother_ql.backends.helpers:Sending instructions to the printer. Total: 103381 bytes.
Traceback (most recent call last):
  File "/home/pi/.local/bin/brother_ql", line 10, in <module>
    sys.exit(cli())
  File "/home/pi/.local/lib/python2.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/.local/lib/python2.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/pi/.local/lib/python2.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/.local/lib/python2.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/.local/lib/python2.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/.local/lib/python2.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/.local/lib/python2.7/site-packages/brother_ql/cli.py", line 147, in print_cmd
    send(instructions=instructions, printer_identifier=printer, backend_identifier=backend, blocking=True)
  File "/home/pi/.local/lib/python2.7/site-packages/brother_ql/backends/helpers.py", line 76, in send
    result = interpret_response(data)
  File "/home/pi/.local/lib/python2.7/site-packages/brother_ql/reader.py", line 159, in interpret_response
    raise NameError("Printer response doesn't start with the usual header (80:20:42)", hex_format(data))
NameError: ("Printer response doesn't start with the usual header (80:20:42)", '61 72 72 61 79 28 27 42 27 2C 20 5B 31 32 38 2C 20 33 32 2C 20 36 36 2C 20 35 32 2C 20 35 36 2C 20 34 38 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 36 32 2C 20 31 31 2C 20 30 2C 20 30 2C 20 34 2C 20 30 2C 20 30 2C 20 31 30 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 31 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 2C 20 30 5D 29')
```

The label will still be printed successfully.

pklaus, 2020-01-30:
There is some issue here, where the received answer is not bytes but an array(). Is this on Python 2.x? The seamingly strange response is actually ASCII representing

```
array('B', [128, 32, 66, 52, 56, 48, 0, 0, 0, 0, 62, 11, 0, 0, 4, 0, 0, 100, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0])
```
So I need to fix something here. Will hopefully be able to do so in the next days. Thanks for reporting!

pklaus, 2020-01-30:
Note to self:

I already analyzed the issue with "Printer response doesn't start with..." in [this comment in #69](https://github.com/pklaus/brother_ql/issues/69#issuecomment-561040150). It's Python 2.x related.

* the bug also appeared in [#75](https://github.com/pklaus/brother_ql/issues/75#issue-545728116), [#71](https://github.com/pklaus/brother_ql/issues/71#issue-520412964) and [more](https://github.com/pklaus/brother_ql/issues?utf8=%E2%9C%93&q=%22Printer+response+doesn%27t+start+with+the+usual+header%22).

henningwerner, 2020-01-30:
Yes it's Python 2.7 I just installed it through pip and not pip3 - this should be my issue.

But would be really cool if you can get this fixed also for Python 2.x.

commit 56cf4394 (2020-02-04): [fix] "response doesn't start with the usual header"

fixes #25, fixes #69, fixes #75, fixes #81

also seen in issues: #56, #71
closed 2020-02-04 by pklaus