pklaus/brother_ql #87: NotImplementedError

gstax, 2020-04-13:
Hi,

Just discovered you package and wanted to give it a try, espcially the --red  option since I have some DK-22251 paper (and just discovered after 3 days, that the linux brother official driver doesn't support printing in red).
I have a QL-820NWB and working from a Debian stable box.

To install, I ran the following command:

`pip install --upgrade https://github.com/pklaus/brother_ql/archive/master.zip`

Then I exported `~/.local/bin` to my PATH. I also
`export BROTHER_QL_MODEL=QL-820NWB`
`export BROTHER_QL_PRINTER=tcp://192.168.1.139`

The printer is being reached via wifi at that address (ping works).

Then I issued:

`brother_ql -b network --debug discover
Traceback (most recent call last):
  File "/home/me/.local/bin/brother_ql", line 10, in <module>
    sys.exit(cli())
  File "/home/me/.local/lib/python2.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/me/.local/lib/python2.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/me/.local/lib/python2.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/me/.local/lib/python2.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/me/.local/lib/python2.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/me/.local/lib/python2.7/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/me/.local/lib/python2.7/site-packages/brother_ql/cli.py", line 47, in discover
    discover_and_list_available_devices(backend)
  File "/home/me/.local/lib/python2.7/site-packages/brother_ql/cli.py", line 51, in discover_and_list_available_devices
    available_devices = discover(backend_identifier=backend)
  File "/home/me/.local/lib/python2.7/site-packages/brother_ql/backends/helpers.py", line 23, in discover
    available_devices = list_available_devices()
  File "/home/me/.local/lib/python2.7/site-packages/brother_ql/backends/network.py", line 25, in list_available_devices
    raise NotImplementedError()
NotImplementedError
`

Since I don't have any Python skills, I lost here.

Help would be very much appreciated.

Thanks

niftylettuce, 2020-04-29:
👍

mattj23, 2020-10-15:
Looking at the file in question (`brother_ql/backends/network.py`) it looks like network discovery hasn't been implemented, but you shouldn't need ti since you know the printer's address already. Have you tried sending something directly to the printer?  I have an 810W and it works fine with the red/black "ink".

gstax, 2020-10-15:
Hi, 

Thanks for you reply.

In fact I didn't want to loose more time so I bought black ink...

Good to know that it works with a 810W.

I'll give it a try once I have some free time.