pklaus/brother_ql #90: Replace media error for brother ql 800 with 12mm labels.

miguelmirandaa, 2020-06-19:
Hello everyone, 

I am having some trouble printing 12mm labels with brother QL-800.
I already printed on 29mm labels endless, as well as 29x90, without any problem. 

I am trying to print an image of size 106*106 pixels(which is the correct size according to: brother_ql_info list-label-sizes

I am using this command to print: `brother_ql -p /dev/usb/lp0 --model QL-800 print -l 12 qr.png`

And I am getting this errors:
`deprecation warning: brother_ql.devicedependent is deprecated and will be removed in a future release
INFO:brother_ql.backends.helpers:Sending instructions to the printer. Total: 10102 bytes.
WARNING:brother_ql.backends.helpers:'printing completed' status not received.
WARNING:brother_ql.backends.helpers:'waiting to receive' status not received.
WARNING:brother_ql.backends.helpers:Printing potentially not successful?
`

michalmph, 2020-06-19:
Deprecation warning is a normal and expected message for the brother_ql.devicedependent library. 

It looks like the printer does not return any statuses after the print request was sent - so the print request either does not reach the printer, or is not understood by it, or something else happened. 
Could you describe the exact procedure that produces this error, or any other details that could help identify your problem?

- Does this happen on printer cold boot with the 12mm tape inside? 
- Do you change the label tapes in the printer while it's powered on, and get this result only when the 12mm tape is in place? 
- Did you successfully print on that particular 12mm tape using PTouch editor, to make sure the tape itself is not a problem? 
- Did you make sure the tape is inserted correctly, and the printer buttons work?
- Do you have any other printers that could occupy /dev/usb/lp0 at the time the request is being sent? 
- Is the "Editor Lite" mode turned off on the printer?

miguelmirandaa, 2020-06-19:
Well, to summarize:

- Does this happen on printer cold boot with the 12mm tape inside? Do you change the label tapes in the printer while it's powered on, and get this result only when the 12mm tape is in place?
- - Yes, I switched the roll with the printer on and off and got the same results.

- Did you successfully print on that particular 12mm tape using PTouch editor, to make sure the tape itself is not a problem?
- - No, I will try it.

-  Did you make sure the tape is inserted correctly, and the printer buttons work?
- - Yes I used the buttons, everything work fine.

- Do you have any other printers that could occupy /dev/usb/lp0 at the time the request is being sent? 
- - No, I only have Brother QL-800

- Is the "Editor Lite" mode turned off on the printer?
- - Yes it is.

If I change the tape to 29mm and execute the same script:
`brother_ql -p /dev/usb/lp0 --model QL-800 print -l 29 qr.png `
It immediately starts working