pklaus/brother_ql #91: Support for red and black printing?

ayjayt, 2020-08-29:
Hey @pklaus , thanks for all the incredible work. It works very well.

I noticed that I can print in red _or_ black depending on which `-l 62red` (red) or -l 62` (black) I set.

I am not very familiar with the raster command set yet, and I am not sure if I will take a look, but from your knowledge do you think it is **feasible** to support red and black printing at the same time?

arnotixe, 2021-01-09:
worksforme: The first label I ever printed with my new QL800 was black and red, and was printed like this:

```
export BROTHER_QL_PRINTER=/dev/usb/lp3
export BROTHER_QL_MODEL=QL-800
brother_ql print -l 62 --red nutella.jpg
```
(the below nutella.jpg I got from google search, resized to 696px wide to fits 62mm labels nicely. But otherwise no editing was done to colors or anything)

[image]

Maybe it would be a good idea to include a demo red/black image in the repo @pklaus ? :) I can confirm the above works great on Q800 anyways

brother_ql rocks! I had to print 5 test labels in windows before I got the settings right (and had to install the drivers…). But on linux _literally_ the first print was a 100% success. Without drivers!

zachchao, 2021-08-04:
The addition of --red solved this for me, should this issue be closed?