pklaus/brother_ql #92: Unable to print with stock 62mm endless label with QL-800

dvdfreitag, 2020-09-22:
Using this command to print:
```
brother_ql -p /dev/usb/lp0 -m QL-800 print -l 62 image.png
```

I get:
```
INFO:brother_ql.backends.helpers:Sending instructions to the printer. Total: 18658 bytes.
WARNING:brother_ql.backends.helpers:'printing completed' status not received.
WARNING:brother_ql.backends.helpers:'waiting to receive' status not received.
WARNING:brother_ql.backends.helpers:Printing potentially not successful?
```

- Happens on cold boot with 62mm roll installed
- Tape is inserted correctly. Feed and Cut buttons work.
- "Editor Lite" mode is disabled
- No other printers are connected to the machine.

I have tried 50, 54, and 62 for label sizes. The printer sits with a red flashing power LED. I have yet to successfully print a single label.

dvdfreitag, 2020-09-22:
Additionally, the following python script produces the same output, no matter what label size I choose.
```python
#!/usr/bin/env python
from reportlab.graphics.shapes import Drawing, String
from reportlab.graphics import renderPM

from brother_ql.devicedependent import label_type_specs
from brother_ql import BrotherQLRaster, create_label
from brother_ql.backends.helpers import send

PRINTER = 'usb://0x04f9:0x209b'
LABEL_NAME = '62x100'
DPI_600 = False

width = label_type_specs[LABEL_NAME]['dots_printable'][1]
height = label_type_specs[LABEL_NAME]['dots_printable'][0]

d = Drawing(width, height)
d.add(String(15, height-300, 'TEST', fontSize=300, fontName='Helvetica'))
d.add(String(15, height-500, '300dpi', fontSize=100, fontName='Helvetica'))

qlr = BrotherQLRaster('QL-800')
create_label(qlr, renderPM.drawToPIL(d), LABEL_NAME, cut=True, rotate=90, dpi_600=False)
send(qlr.data, PRINTER)
```

dvdfreitag, 2020-09-22:
I figured it out. These labels must be printed with the red parameter, which seems extremely unintuitive.
```python
create_label(qlr, renderPM.drawToPIL(d), LABEL_NAME, cut=True, red=True, rotate=90, dpi_600=False)
```

closed 2020-09-22 by dvdfreitag

konz-rs, 2022-01-31:
Hey @dvdfreitag ! Many thanks for writing this down! I wasted quite a few hours on this and was about to go back to the Windows printer driver 🤢 when I found your report. Works fine now! :-)

FriedrichFroebel, 2022-01-31:
This actually is documented in the options of `brother_ql print --help`:

```
   --red                          Create a label to be printed on
                                  black/red/white tape (only with QL-8xx
                                  series on DK-22251 labels). You must use
                                  this option when printing on black/red tape,
                                  even when not printing red.
```

But I admit that this might be easy to miss if you are not actually looking for printing with black/red tape.