pklaus/brother_ql #97: unable to function on a Pi3B with brother QL-570

zacharyjordan23, 2020-11-08:
im getting premission denied when trying to print.  and when I run /dev/usb/lp0 , it also says premission denied, even though I have modified premission on the lp0 file to change the owner to me(the current user)

when trying to print : 
   raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)

when running /dev/usb/lp0   : 
bash: /dev/usb/lp0: Permission denied

PLEASE HELP/ GIVE ME A RECOMMENDATION OF TROUBLESHOOTING STEPS, IVE TRIED EVERYTHING

kumekay, 2020-11-29:
Likely it's something with your udev rules.
Try to create a file `/etc/udev/rules.d/99-usb.rules` 
with content like:

```
SUBSYSTEM=="usb", ATTR{idVendor}=="04f9", ATTR{idProduct}=="****", MODE="0666"
```

where you should replace `****` with the actual product id. You may run `lsusb` to find it, there will be a line like

```
Bus 005 Device 017: ID 04f9:20c0 Brother Industries, Ltd ...
```

Then run `sudo udev trigger` or reboot the device to apply new rules.

kumekay, 2020-12-04:
A correction to instructions above, they are valid if you are using a printer with pyusb with an address like
`usb://0x04f9:0x20c0/000G9Z495858` 
where 

- `0x04f9` - vendor id
- `0x20c0`  - product id
- `000G9Z495858` - serial

You can find these numbers by running `lsusb -d 04f9: -v`

```
Bus 003 Device 004: ID 04f9:20c0 Brother Industries, Ltd QL-600
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x04f9 Brother Industries, Ltd
  idProduct          0x20c0
  bcdDevice            1.00
  iManufacturer           1 Brother
  iProduct                2 QL-600
  iSerial                 3 000G9Z495858
  bNumConfigurations      1
```

kumekay, 2020-12-04:
If you want to use device like `/dev/usb/lp0` you should add your user to `lp` group (and then restart the session):
`sudo usermod -a -G lp your_user`