pklaus/brother_ql #108: deprecation warning: brother_ql.devicedependent is deprecated

planetar, 2021-04-03:
I just discovered this *very useful* project and am still finding my ways through it.
From the readme.md I understand that there are legacy command line tools, such as brother_ql_create, brother_ql_print, brother_ql_analyze which still work but are deprecated now. 

I call the program like:
brother_ql print -l 62  $path$file

which prints the label nicely and prompt. However, on the cli I receive a warning with every single call:

deprecation warning: brother_ql.devicedependent is deprecated and will be removed in a future release

This sort of irritates me, is  'brother_ql' deprecated too? How can I call the program and avoid getting those warnings?

raspbian on raspberry 2 
brother_ql --version
deprecation warning: brother_ql.devicedependent is deprecated and will be removed in a future release
brother_ql, version 0.9.4

fnellen, 2022-05-05:
I am also getting this message while running:
`brother_ql -b pyusb -m 'QL-570' -p usb://0x04f9:0x2028 print -l '62x29' voucher.png`
Is it an error or only a warning?

loeffelandi, 2026-10-17:
Thought I'd share a data point: I'm building a Linux GUI on top of brother_ql 0.9.4 (QLInux, GPL-3.0-or-later, https://github.com/loeffelandi/QLInux). We pin 0.9.4 because master removes `brother_ql.devicedependent`, which breaks our imports. We also apply the `Image.ANTIALIAS` -> `Image.LANCZOS` change from #168 on install, since 0.9.4 still uses it.

Not a request, just so people searching for a GUI on top of this library know it exists.